riscv-software-src/riscv-tests #5: Compilation failure for sb test

zhemao, 2015-12-14:
After updating to the latest commit of riscv-tests, I get the following compilation failure.

    riscv64-unknown-elf-gcc  -DENTROPY=188850 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -std=gnu99 -O2 -I./../env/v -I./macros/scalar -T./../env/v/link.ld ./../env/v/entry.S ./../env/v/vm.c rv64ui/sb.S -lc -o rv64ui-v-sb
    rv64ui/sb.S: Assembler messages:
    rv64ui/sb.S:63: Error: illegal operands `la x1,0xffffffffffffffdd'
    rv64ui/sb.S:64: Error: illegal operands `la x1,0xffffffffffffffcd'
    rv64ui/sb.S:65: Error: illegal operands `la x1,0xffffffffffffffcc'
    rv64ui/sb.S:66: Error: illegal operands `la x1,0xffffffffffffffbc'
    rv64ui/sb.S:67: Error: illegal operands `la x1,0xffffffffffffffbb'
    rv64ui/sb.S:68: Error: illegal operands `la x1,0xffffffffffffffab'
    rv64ui/sb.S:70: Error: illegal operands `la x1,0x33'
    rv64ui/sb.S:71: Error: illegal operands `la x1,0x23'
    rv64ui/sb.S:72: Error: illegal operands `la x1,0x22'
    rv64ui/sb.S:73: Error: illegal operands `la x1,0x12'
    rv64ui/sb.S:74: Error: illegal operands `la x1,0x11'
    rv64ui/sb.S:75: Error: illegal operands `la x1,0x01'

I've already tried updating my compiler, but I still get this error message.

colinschmidt, 2015-12-14:
@palmer-dabbelt I think your change https://github.com/riscv/riscv-gnu-toolchain/commit/4fcaa1c95ad33b94b9be9234ab2e79e2e9aad9bb is the cause of this issue,

I see two solutions:
Add a non-symbolic 'la'
If these 'la' instructions always load a constant change them to 'li'

zhemao, 2015-12-14:
Changing "la" to "li" in the macro seems to work.

zhemao, 2015-12-14:
Pushed the fix to master.

closed 2015-12-14 by zhemao

aswaterman, 2015-12-14:
I think la being symbolic-only is ok.

colinschmidt, 2015-12-14:
Cool, I agree.