riscv-software-src/riscv-tests #13: Error: Instruction csrr requires absolute expression

AngelTerrones, 2016-05-04:
Hi. I'm trying to build the tests, but i'm getting the following error (using `master` branch of this repo):

```
home/angelterrones/Sources/RISCV/riscv-tests/benchmarks/common/syscalls.c: Assembler messages:
/home/angelterrones/Sources/RISCV/riscv-tests/benchmarks/common/syscalls.c:57: Error: Instruction csrr requires absolute expression
/home/angelterrones/Sources/RISCV/riscv-tests/benchmarks/common/syscalls.c:57: Error: Instruction csrr requires absolute expression
/home/angelterrones/Sources/RISCV/riscv-tests/benchmarks/Makefile:83: recipe for target 'syscalls.o' failed
make[1]: *** [syscalls.o] Error 1
make[1]: Leaving directory '/home/angelterrones/Sources/RISCV/riscv-tests/benchmarks'
Makefile:16: recipe for target 'benchmarks' failed
make: *** [benchmarks] Error 2
```

I have updated the `riscv-tools` following the instructions from the main repo (fresh clone):

```
$ git submodule update --init --recursive
$ export RISCV=/path/to/install/riscv/toolchain
$ ./build.sh
```

Thanks.

aswaterman, 2016-05-04:
In general, you need to use a consistent set of riscv-tools, since the components are under development and change rapidly.  Active development is occurring on the priv-1.9 branch of riscv-tools.

closed 2016-05-04 by aswaterman

AngelTerrones, 2016-05-06:
Ok then. I expected  `master` being the stable branch (priv v1.7).
Thanks.