riscv-software-src/riscv-tests #23: requirements.txt for Python scripts?

mwachs5, 2016-08-23:
The gdbserver.py requires some non-standard Python packages, notably pexpect. Can we add a requirements.txt file to aid in automated testing with gdbserver?

richardxia, 2016-08-23:
In addition, can we change the shebang line of [`gdbserver.py`](https://github.com/riscv/riscv-tests/blob/c8c14a5c4f29655465296fd8e3b2371c6899f604/debug/gdbserver.py#L1) to be `#!/usr/bin/env python`? The current value of `#!/usr/bin/python` prevents the use of `virtualenv`, since activating a virtualenv will change the path of Python to a special one within the virtualenv that ensures that packages within the virtualenv are in the load path.

`#!/usr/bin/env python` is also the standard shebang line used for Python scripts, since Python is not guaranteed to always be installed to `/usr/bin/python`: http://stackoverflow.com/a/19305076

timsifive, 2016-08-23:
I've changed the shebang line as richardxia requested.

Megan, can you create the appropriate requirements.txt, or let me know how I can test that one I generate works?

richardxia, 2016-08-23:
I can submit a pull request with the requirements.txt.

mwachs5, 2017-02-12:
@richardxia we can close this right?

richardxia, 2017-02-13:
Yes, we can close it now. I don't have permission to, though, so someone else needs to do it.

closed 2017-02-13 by timsifive