riscv-software-src/riscv-tests #25: isa tests does not have start instructions

alexmechanic, 2016-08-25:
Correct me if I'm wrong, but i cloned this repo and built tests using your toolchain.
But some found some issues:
1. riscv-qemu faults running these tests (segfault or memory allocation fault). Followed the instruction [(on your site)](https://riscv.org/software-tools/riscv-qemu/)
2. starting pc in the test header (ex., rv32mi-p-csr) is set to 80002000, but there is no code in this place (section .tohost)
Can you explain me how to fix this?

susi655, 2016-09-02:
Can you give me a clarity regarding how this riscv-tests useful?
 or how you are planning to use?

alexmechanic, 2016-09-03:
I wanted to see how riscv arch works like regs initial state, branching etc. Why are you asking?

susi655, 2016-09-03:
I wanted to know how riscv-tests work? I have hardware in FPGA tools, I was really struggling to test in a proper way? That's why i asked!

alexmechanic, 2016-09-05:
Well, I have some simulator and I want to test riscv integration for future FPGA integration