riscv-software-src/riscv-tests #30: separate make target for running debug tests?

aswaterman, 2016-09-19:
We probably should keep it so that make only builds the tests, and make check executes them.  Main reason is to speed up building the tools, which is a significant bottleneck for regression testing.

aswaterman, 2016-09-20:
To clarify, I think the regression tests for riscv-tools _should_ make check, but other things that depend on riscv-tools (e.g. rocket-chip) should not.

timsifive, 2016-09-20:
I don't exactly follow this. There is a separate make target for the debug tests, 'make debug'.

Is this a change that needs to be made in the rocket repo?

aswaterman, 2016-09-20:
The issue is that "make" depends on "make debug," which, in addition to building the debug tests, executes them.  By contrast, "make" only builds the ISA tests, rather than executing them.

timsifive, 2016-09-20:
I see. So how do they get executed? I'd like for the debug tests to get executed at least as part of the riscv-tools travis build.

aswaterman, 2016-09-20:
I agree.  I think it's just a matter of modifying riscv-tools/.travis.yml to do something like

    script:
      - ./build.sh
      - cd riscv-tools/build
      - make debug-check

timsifive, 2016-09-21:
OK. I don't have access to riscv-tools, and there doesn't seem to be a way to test travis changes without actually pushing to the repo. Can you make this change, or give me access?

aswaterman, 2016-09-21:
I'll give it a shot

timsifive, 2016-09-26:
Looks like it's not working because of a missing dependency. Adding python-pexpect to the list of packages should fix it.
(Note that there used to be a bug in the Makefile that would cause 'make' to return even if the tests failed, so it could be that they have never been run successfully at all in travis.)

aswaterman, 2016-09-27:
Makes sense, I'll give it a shot.