riscv-software-src/riscv-tests #33: Load and Store operations

alexmechanic, 2016-10-12:
Ex., test **rv32ui-p-sh**, section *800001c4 <test_9>*:
```
800001c4:	00002097          	auipc	ra,0x2
800001c8:	e4a08093          	addi	ra,ra,-438 # 8000200e <tdat8>
800001cc:	ffffa137          	lui	sp,0xffffa
800001d0:	00a10113          	addi	sp,sp,10 # ffffa00a <_end+0x7fff7fea>
800001d4:	00209023          	sh	sp,0(ra)
800001d8:	00009183          	lh	gp,0(ra)
800001dc:	ffffaeb7          	lui	t4,0xffffa
800001e0:	00ae8e93          	addi	t4,t4,10 # ffffa00a <_end+0x7fff7fea>
800001e4:	00900e13          	li	t3,9
800001e8:	35d19663          	bne	gp,t4,80000534 <fail>
```
Test supposed to store 16 lower bits of *0xffffa00a* to memory (which is *0xa00a*) as it is said in your manual:
`The SW, SH, and SB instructions store 32-bit, 16-bit, and 8-bit values from the low bits of register rs2 to memory.`

Then test supposed to load 16-bit value from memory, which is the same *0xa00a*, but then test checks the value for ** ffffa00a**. 

Why is that?

vanjoe, 2016-10-12:
The lh instruction sign extends to the upper bits. If the instruction at
1d8 was lha, it would zero extend the value.

alexmechanic, 2016-10-12:
@vanjoe You mean, if lh loads 16-bit value and its MSB (val[15]) is 1, then it sign-extends it to upper 16 bits?

vanjoe, 2016-10-12:
Yup.

alexmechanic, 2016-10-14:
No problem here

closed 2016-10-14 by alexmechanic