riscv-software-src/riscv-tests #36: TriggerLoadAddress read watch breakpoint issue

rboyinee, 2016-11-17:
I am trying to run debug tests in simulation mode on ROCKET-CHIP in Redhat linux machine.

When I am setting a read watch breakpoint as mentioned in "TriggerLoadAddress" test as mentioned below

(gdb) rwatch *((&data)+1)
Hardware read watchpoint 3: *((&data)+1)

I am getting the following error, 
(gdb) c
Continuing.
keep_alive() was not invoked in the 1000ms timelimit. GDB alive packet not sent! (2943). Workaround: increase "set remotetimeout" in GDB
"dbus_scan failed jtag scan
Timed out waiting for debug int to clear.
Debug interrupt didn't clear"

Are all the debug tests work on ROCKET-CHIP hardware in simulation mode or is there anything I am missing in my environment.

timsifive, 2016-11-17:
I don't know anything about Rocket chip, but if you run openocd with "-d"
and send me the output I can try and figure out what happened.

Tim

rboyinee, 2016-11-18:
I am attaching the openocd output.
[openocd.zip](https://github.com/riscv/riscv-tests/files/599904/openocd.zip)

timsifive, 2016-11-18:
Thanks. I don't know why this is failing, but I can describe in a little
more detail what happens right before it fails:
1. OpenOCD configures CSRs for a read watchpoint at 0x80000554. (I don't
recall if I've actually tested watchpoints on RV64, but it ought to work.
Certainly the worst case should be benign.)
2. OpenOCD performs a single step. A lot of GPRs have the
value 0xdcfc5eb9dcfc5eb9. I assume that's expected?
3. OpenOCD sets a software breakpoint at 0x80000380
4. OpenOCD starts running.
5. Then something mysterious happens, and JTAG scanning stops working. I've
never seen this error. It indicates that at some low level OpenOCD was
unable to shift bits. I don't think there's anything that the Rocket
simulator can do to cause this. Is there a time limit to how long the
simulator-JTAG thing (jtag_vpi?) runs?

Tim