riscv-software-src/riscv-tests #39: How do I use Debug Test for Rocket-Chip core in simulation

colin4124, 2016-12-29:
There are two targets in simulation (freedom-e300-sim and freedom-u500-sim) https://github.com/riscv/riscv-tests/tree/master/debug 

**What about testing debug for default rocket chip in simulation?** 

I have a try as described below：

####  **1. Add `WithJtagDTM` to Rocket Chip default configuration：** 
appending code as following to `rocket-chip/src/main/scala/rocketchip/Configs.scala`, witch I copy from https://github.com/sifive/freedom/blob/master/src/main/scala/unleashed/u500vc707devkit/Configs.scala#L9
```
class DefaultFreedomUConfig extends Config(
  new WithJtagDTM ++ new BaseConfig
)
```

#### **2. Build the VCS simulator with `DefaultFreedomUConfig` configuration:**
`$ make CONFIG=DefaultFreedomUConfig`

#### **3. Run Debug Test:** 
I have built riscv tools using 64 bits, so I use **freedom-u500-sim** target.
`$ python2.7 gdbserver.py --freedom-u500-sim --run rocket-chip/vsim/simv-rocketchip-DefaultFreedomUConfig`

I found that it runs into infinite loop in [testlib.VcsSim](https://github.com/riscv/riscv-tests/blob/master/debug/testlib.py#L93)  
``` python
        while not done:
            line = listenfile.readline()
            if not line:
                time.sleep(1)
            match = re.match(r"^Listening on port (\d+)$", line)
            if match:
                done = True
                self.port = int(match.group(1))
                os.environ['JTAG_VPI_PORT'] = str(self.port)
```

In `simv.log`:
```
+ rocket-chip/vsim/simv-rocketchip-DefaultFreedomUConfig +jtag_vpi_enable
Chronologic VCS simulator copyright 1991-2014
Contains Synopsys proprietary information.
Compiler version I-2014.03_Full64; Runtime version I-2014.03_Full64;  Dec 29 07:55 2016
```

I think that `simv-rocketchip-DefaultFreedomUConfig` never output strings like "Listening on port", so it would be in infinite loop.

I wonder if I missing some arguments in `$ python2.7 gdbserver.py --freedom-u500-sim --run rocket-chip/vsim/simv-rocketchip-DefaultFreedomUConfig` or  `simv-rocketchip-DefaultFreedomUConfig` maybe take a long time so I just wait for it?

richardxia, 2016-12-30:
Even though you're using the correct Config with the `WithJtagDTM` trait, you're still using the default `ExampleTop` which does not mix in the necessary Periphery traits in order to use the JTAG debug module. In addition, the default `TestHarness` also does not instantiate the JTAG VPI module.

You'll want to modify `ExampleRocketTop`, `ExampleRocketTopBundle`, and `ExampleRocketTopModule` to use `PeripheryJTAG*` instead of `PeripheryDTM*`. You'll also want to modify `TestHarness` to instantiate a `JTAGVPI` instead of a `SimDTM`.

colin4124, 2016-12-30:
@richardxia Thinks for your answer.

If I have simulated u500 and e300 (https://github.com/sifive/freedom) through the VLSI tools, can I just run below command to test Debug?
`python2.7 gdbserver.py --freedom-u500-sim --run path/to/simv-rocketchip-U500VC707DevKitConfig`

But I still don't know how to simulated u500 and e300 by vcs yet. I have open an issue about that: https://github.com/sifive/freedom/issues/11

richardxia, 2017-01-03:
Yes, you should be able to run that command to test the debug module. Note that the tests are written against a C simulator (spike), so you may need to increase the test timeouts in order to successfully test against an RTL simulator.

mwachs5, 2018-04-19:
This is now done in rocket-chip's regression suite and can be used as an example, so I am closing it here.

closed 2018-04-19 by mwachs5