riscv-software-src/riscv-tests #40: "Error: value of N too large for field" errors on macOS Sierra

brettcannon, 2017-01-04:
I tried to build on macOS Sierra today with an updated GNU toolchain (all other RISC-V code was from riscv/homebrew-riscv) and I got this error:
```
riscv64-unknown-elf-gcc -march=rv32g -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -I/Users/brettcannon/Repositories/harmony/riscv-tests/isa/../env/p -I/Users/brettcannon/Repositories/harmony/riscv-tests/isa/macros/scalar -T/Users/brettcannon/Repositories/harmony/riscv-tests/isa/../env/p/link.ld rv32ui/sll.S -o rv32ui-p-sll
rv32ui/../rv64ui/sll.S: Assembler messages:
rv32ui/../rv64ui/sll.S:34: Error: value of 71144870016 too large for field of 4 bytes at 568
rv32ui/../rv64ui/sll.S:34: Error: value of 71144870016 too large for field of 4 bytes at 572
rv32ui/../rv64ui/sll.S:35: Error: value of 9106543362048 too large for field of 4 bytes at 600
rv32ui/../rv64ui/sll.S:36: Error: value of 1193612851550355456 too large for field of 4 bytes at 628
rv32ui/../rv64ui/sll.S:42: Error: value of 71144870016 too large for field of 4 bytes at 720
rv32ui/../rv64ui/sll.S:42: Error: value of 71144870016 too large for field of 4 bytes at 724
rv32ui/../rv64ui/sll.S:43: Error: value of 9106543362048 too large for field of 4 bytes at 752
make[1]: *** [rv32ui-p-sll] Error 1
make: *** [isa] Error 2
```

Is this something broken on my machine or is this a bug while running under macOS?

aswaterman, 2017-01-04:
In general, you should only use a consistent set of riscv-tools, rather than selectively updating individual repos.  However, this happens to have been fixed earlier today (pull riscv-tests master).

closed 2017-01-04 by aswaterman

brettcannon, 2017-01-04:
Yeah, I ended up in the odd tools state when helping with riscv/homebrew-riscv to figure out if the Homebrew bottles were outdated.

Thanks for the info!