riscv-software-src/riscv-tests #42: ./gdbserver.py --spike64 --cmd $RISCV/bin/spike fails

riscCoder, 2017-02-20:
I am missing module 'pexpect' when I invoke the following.

> ./gdbserver.py --spike64 --cmd $RISCV/bin/spike

testlib.py line 9 can not import 'pexpect'.  I could not find pexpect.* in my RISCV installation.

riscCoder, 2017-02-20:
'pexpect' is found in a python system module.  I found 'python-pexpect' in my Linux package manager and installed.  But now I have errors with the python libraries in "/usr/lib64/python2.7/subprocess.py".  I am running on a relatively fresh Linux installation(Feb 2017).  So, it appears these errors are version related problems with python.

timsifive, 2017-02-20:
What is the full output of the program when you execute it?

riscCoder, 2017-02-20:
Here is the complete screen capture

> ./gdbserver.py --spike32 --cmd ../bin/spike
Running ExamineTarget ...
Traceback (most recent call last):
  File "./gdbserver.py", line 701, in <module>
    sys.exit(main())
  File "./gdbserver.py", line 693, in main
    return testlib.run_all_tests(module, target, parsed)
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/testlib.py", line 350, in run_all_tests
    result = instance.run()
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/testlib.py", line 456, in run
    self.classSetup()
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/testlib.py", line 495, in classSetup
    BaseTest.classSetup(self)
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/testlib.py", line 430, in classSetup
    self.server = self.target.server()
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/targets.py", line 85, in server
    return testlib.Spike(self.cmd, halted=True, xlen=32)
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/testlib.py", line 81, in __init__
    stdout=logfile, stderr=logfile)
  File "/usr/lib64/python2.7/subprocess.py", line 711, in __init__
    errread, errwrite)
  File "/usr/lib64/python2.7/subprocess.py", line 1343, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory

timsifive, 2017-02-20:
The problem is that it cannot find the spike executable. Is `../bin/spike` the right path?

riscCoder, 2017-02-21:
Yea let me try that again.

echo $PWD
/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug
dave@linux:~/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug> ../../bin/spike -h
usage: spike [host options] <target program> [target options]
Host Options:
  -p<n>                 Simulate <n> processors [default 1]
  -m<n>                 Provide <n> MiB of target memory [default 4096]
  -d                    Interactive debug mode
  -g                    Track histogram of PCs
  -l                    Generate a log of execution
  -h                    Print this help message
  -H                 Start halted, allowing a debugger to connect
  --isa=<name>          RISC-V ISA string [default RV64IMAFDC]
  --ic=<S>:<W>:<B>      Instantiate a cache model with S sets,
  --dc=<S>:<W>:<B>        W ways, and B-byte blocks (with S and
  --l2=<S>:<W>:<B>        B both powers of 2).
  --extension=<name>    Specify RoCC Extension
  --extlib=<name>       Shared library to load
  --gdb-port=<port>  Listen on <port> for gdb to connect
  --dump-config-string  Print platform configuration string and exit
dave@linux:~/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug> echo $RISCV
/home/dave/MyProjects/RiscV/build/riscv-tools
dave@linux:~/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug> ./gdbserver.py --spike32 --cmd ../../bin/spike
Running ExamineTarget ... RV32ACDFIMSU pass in 1.14s
Running DebugBreakpoint ...
Traceback (most recent call last):
  File "./gdbserver.py", line 701, in <module>
    sys.exit(main())
  File "./gdbserver.py", line 693, in main
    return testlib.run_all_tests(module, target, parsed)
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/testlib.py", line 350, in run_all_tests
    result = instance.run()
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/testlib.py", line 456, in run
    self.classSetup()
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/testlib.py", line 495, in classSetup
    BaseTest.classSetup(self)
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/testlib.py", line 428, in classSetup
    self.compile()
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/testlib.py", line 424, in compile
    self.target.compile(*compile_args)
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/targets.py", line 57, in compile
    xlen=self.xlen)
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/testlib.py", line 31, in compile
    stderr=subprocess.PIPE)
  File "/usr/lib64/python2.7/subprocess.py", line 711, in __init__
    errread, errwrite)
  File "/usr/lib64/python2.7/subprocess.py", line 1343, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
dave@linux:~/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug> 

Just to be sure I'll invoke with $RISC env variable in case it is needed to find gdb.

dave@linux:~/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug> ./gdbserver.py --spike32 --cmd $RISCV/bin/spike

Running ExamineTarget ... RV32ACDFIMSU pass in 1.14s
Running DebugBreakpoint ...
Traceback (most recent call last):
  File "./gdbserver.py", line 701, in <module>
    sys.exit(main())
  File "./gdbserver.py", line 693, in main
    return testlib.run_all_tests(module, target, parsed)
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/testlib.py", line 350, in run_all_tests
    result = instance.run()
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/testlib.py", line 456, in run
    self.classSetup()
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/testlib.py", line 495, in classSetup
    BaseTest.classSetup(self)
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/testlib.py", line 428, in classSetup
    self.compile()
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/testlib.py", line 424, in compile
    self.target.compile(*compile_args)
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/targets.py", line 57, in compile
    xlen=self.xlen)
  File "/home/dave/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug/testlib.py", line 31, in compile
    stderr=subprocess.PIPE)
  File "/usr/lib64/python2.7/subprocess.py", line 711, in __init__
    errread, errwrite)
  File "/usr/lib64/python2.7/subprocess.py", line 1343, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory

timsifive, 2017-02-21:
> usage: spike [host options] [target options]
> -p Simulate processors [default 1]
> -m Provide MiB of target memory [default 4096]
> --isa= RISC-V ISA string [default RV64IMAFDC]
> * Instantiate a cache model with S sets, --dc=:: W ways, and B-byte blocks
> (with S and --l2=:: B both powers of 2). --extension= Specify RoCC
> Extension --extlib= Shared library to load --gdb-port= Listen on for gdb to
> recent call last): File "./gdbserver.py", line 701, in sys.exit(main())

Now the problem is that it can't find the compiler. It's probably looking
for riscv32-unknown-elf-gcc.

Tim

riscCoder, 2017-02-21:
I have been successfully building code with $RISCV/bin/riscv64-unknown-elf-gcc.  Are you saying that 
these python routines require a rsicv32 toolchain?

timsifive, 2017-02-21:
Yes, to build 32-bit binaries they invoke riscv32-...

Tim

riscCoder, 2017-02-21:
This makes no sense to me. A 64 bit spike is built along with the 64 bit RISCV toolchain.  I've done that and it appears to be functional.  But to test gdb using this 64 bit spike - you would need to build the 32 bit RISCV tools?  

And gdb used with 64 bit spike - is it 32 or 64 bit?

timsifive, 2017-02-21:
You're welcome to change it to be more flexible. Looks like it always runs
the 64-bit gdb.

riscCoder, 2017-02-21:
I just realized I invoked with the --spike32 option.  

Here is the result when it is invoked with --spike64 option.  Sorry about the screw up.
Operator error aside - the only issue was I needed to add the package 'python-pexpect' and 
this thing should have executed with no issues.  Thanks for the help.

dave@linux:~/MyProjects/RiscV/build/riscv-tools/riscv-tests/debug> ./gdbserver.py --spike64 --cmd $RISCV/bin/spike        
Running ExamineTarget ... RV64ACDFIMSU pass in 1.17s
Running DebugBreakpoint ... pass in 3.48s
Running DebugChangeString ... pass in 3.92s
Running DebugCompareSections ... pass in 1.81s
Running DebugExit ...^[[24~ pass in 1.86s
Running DebugFunctionCall ... pass in 3.99s
Running DebugSymbols ... pass in 2.43s
Running DebugTurbostep ... pass in 30.17s
Running DownloadTest ... pass in 14.43s
Running Hwbp1 ... pass in 3.33s
Running Hwbp2 ... pass in 3.91s
Running InstantChangePc ... pass in 2.24s
Running InstantHaltTest ... pass in 1.20s
Running MemTest16 ... pass in 1.74s
Running MemTest32 ... pass in 1.74s
Running MemTest64 ... pass in 1.74s
Running MemTest8 ... pass in 1.74s
Running MemTestBlock ... pass in 3.39s
Running MemTestReadInvalid ... pass in 1.28s
Running MemTestWriteInvalid ... pass in 1.32s
Running MprvTest ... pass in 2.12s
Running PrivChange ... pass in 2.59s
Running PrivRw ... pass in 2.94s
Running Registers ... pass in 8.21s
Running SimpleF18Test ... pass in 2.64s
Running SimpleS0Test ... pass in 2.53s
Running SimpleS1Test ... pass in 2.53s
Running SimpleT0Test ... pass in 2.52s
Running SimpleT1Test ... pass in 2.52s
Running StepTest ... pass in 4.50s
Running TriggerDmode ... pass in 5.04s
Running TriggerExecuteInstant ... pass in 2.26s
Running TriggerLoadAddress ... pass in 2.80s
Running TriggerLoadAddressInstant ... pass in 2.98s
Running TriggerStoreAddress ... pass in 2.80s
Running TriggerStoreAddressInstant ... pass in 2.99s
Running UserInterrupt ... pass in 3.34s
Running WriteCsrs ... pass in 3.53s
Running WriteGprs ... pass in 12.06s
::::::::[ ran 39 tests in 158s ]::::::::
39 tests returned pass

closed 2017-02-22 by riscCoder