riscv-software-src/riscv-tests #64: Build fails with ImportError: No module named spike64

jonathanrainer, 2017-08-10:
Tried to install RISC-V Tests on a new VM and got an error during install. Error text reads:

`Traceback (most recent call last):
  File "/opt/riscv-tools/riscv-tests/build/../debug/gdbserver.py", line 787, in <module>
    sys.exit(main())
  File "/opt/riscv-tools/riscv-tests/build/../debug/gdbserver.py", line 772, in main
    target = targets.target(parsed)
  File "/opt/riscv-tools/riscv-tests/debug/targets.py", line 136, in target
    module = importlib.import_module(module_name)
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named spike64
`

Any ideas?

timsifive, 2017-08-10:
This looks like an error that you might (but shouldn't) get when running
the debug tests. I wouldn't expect that even to happen during regular
install.

What command did you run, and what directory did you run it from?

Tim

jonathanrainer, 2017-08-10:
I simply cloned the riscv-tools repo and then ran the build.sh script that's included, everything else seemed to be fine other than generating a lot of warnings. I had some problem with Pexpect but after I installed that it seemed fine till this error appeared

timsifive, 2017-08-10:
I think I figured out what's going on. It should be fixed with the above change.

closed 2017-08-11 by jonathanrainer