riscv-software-src/riscv-tests #75: dhrystone

esolera, 2017-09-14:
I tried to compile dhrystone for 32 bits with:
riscv32-unknown-elf-gcc -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/ri
scv-tools/riscv-tests/build/../benchmarks/../env -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/common -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/median -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/qsort -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/rsort -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/towers -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/vvadd -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/multiply -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/mm -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/dhrystone -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/spmv -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/mt-vvadd -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/mt-matmul -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/pmp -DPREALLOCATE=1 -mcmodel=medany -static -std=gnu99 -O2 -ffast-math -fno-common -fno-builtin-printf -static -nostdlib -nostartfiles -lgcc -T /home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/common/test.ld -o dhrystone.riscv /home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/dhrystone/dhrystone.c /home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/dhrystone/dhrystone_main.c /home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/common/syscalls.c /home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/common/crt.S

But faiol and say me i have this error:
/tmp/cckATb8W.o: In function `.L47':
syscalls.c:(.text+0x13c): undefined reference to `__umoddi3'
/tmp/cckATb8W.o: In function `.L58':
syscalls.c:(.text+0x168): undefined reference to `__udivdi3'
syscalls.c:(.text+0x180): undefined reference to `__umoddi3'
collect2: error: ld returned 1 exit status

aswaterman, 2017-09-14:
Thanks for the bug report.  Can you confirm this commit fixes it: https://github.com/riscv/riscv-tests/commit/e0b295faa65c8e6bae880fc17a53d481ef74402f

esolera, 2017-09-19:
Thanks the commits work well for compiling : median, rsort, qsort, vvadd, multiply, towers and dhrystone

But when a tried to compile mm with : 
riscv32-unknown-elf-gcc -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/../env -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/common -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/median -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/qsort -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/rsort -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/towers -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/vvadd -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/multiply -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/mm -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/dhrystone -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/spmv -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/mt-vvadd -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/mt-matmul -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/pmp -DPREALLOCATE=1 -mcmodel=medany -static -std=gnu99 -O2 -ffast-math -fno-common -fno-builtin-printf -o mm.riscv /home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/mm/mm.c /home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/mm/mm_main.c /home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/common/syscalls.c /home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/common/crt.S -static -nostdlib -nostartfiles -lgcc -T /home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/common/test.ld

I have this error : 
/tmp/ccld0iSb.o: In function `.L5':
mm.c:(.text+0x140): undefined reference to `fma'
mm.c:(.text+0x16c): undefined reference to `fma'
mm.c:(.text+0x198): undefined reference to `fma'
mm.c:(.text+0x1c4): undefined reference to `fma'
/tmp/ccld0iSb.o: In function `.L7':
mm.c:(.text+0x274): undefined reference to `fma'
/tmp/ccld0iSb.o:mm.c:(.text+0xcbc): more undefined references to `fma' follow
collect2: error: ld returned 1 exit status
/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/Makefile:54: recipe for target 'mm.riscv' failed
make[1]: *** [mm.riscv] Error 1
make[1]: Leaving directory '/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/benchmarks'
Makefile:18: recipe for target 'benchmarks' failed
make: *** [benchmarks] Error 2

Thanks for helping

aswaterman, 2017-09-19:
Try 51b823c7ed8391756a83cab854cd4c646dd13acf