riscv-software-src/riscv-tests #81: Error compiling from architecture RV64IMA

esolera, 2017-10-02:
Tried to compile som fir becnhmark test and with this instruction :  
riscv64-unknown-elf-gcc -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/../env -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/common -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/median -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/16math -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/16switch -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/16switchcase -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/32math -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/8math -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/8switch -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/8switchcase -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/fir -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/floatingpointmath -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/matrixmulti -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/qsort -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/rsort -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/towers -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/vvadd -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/multiply -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/mm -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/dhrystone -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/spmv -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/mt-vvadd -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/mt-matmul -I/home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/pmp -DPREALLOCATE=1 -mcmodel=medany -march=rv64ima -mabi=lp64 -static -std=gnu99 -O2 -ffast-math -fno-common -fno-builtin-printf -o fir.riscv /home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/fir/fir.c /home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/common/syscalls.c /home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/common/crt.S -static -nostdlib -nostartfiles -lm -lgcc -T /home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv-tests/build/../benchmarks/common/test.ld

and I ge}ot this error: 
/tmp/cc4S73gG.o: In function `.L8':
fir.c:(.text.startup+0xac): undefined reference to `__floatunsisf'
/tmp/cc4S73gG.o: In function `.L3':
fir.c:(.text.startup+0xb8): undefined reference to `__mulsf3'
fir.c:(.text.startup+0xc4): undefined reference to `__addsf3'
fir.c:(.text.startup+0xdc): undefined reference to `__floatunsisf'
fir.c:(.text.startup+0xec): undefined reference to `__mulsf3'
fir.c:(.text.startup+0xf8): undefined reference to `__addsf3'
collect2: error: ld returned 1 exit status
make[1]: *** [fir.riscv] Error 1
make: *** [benchmarks] Error 2

aswaterman, 2017-10-03:
libgcc is supposed to contain definitions of __addsf3 and __mulsf3, but only if the compiler was built to to assume there is no hardware floating-point.  It sort of looks like you're using an rv64imafd compiler, but trying to build rv64ima code.

jhcloos, 2017-10-03:
Any riscv gcc/binutils ought to be able to target any rv isa.

The current ones seem to be rv32 or rv64 (or rv128?) only, but w/in those sets they should be able to compile to each of the possible sets of extensions.

If the current config for gcc fails to support that, that ought to be handled as a bug.

esolera, 2017-10-03:
how i can build rv64ima compiler ?????

aswaterman, 2017-10-03:
@jhcloos It's not a bug.  You need to enable multilibs for that to really work, because you need to generate a libgcc for each ISA.  If you enable multilibs, it works as you'd hope.  This is the same for other architectures, too.

aswaterman, 2017-10-03:
@esolera you can try adding ```--with-arch=rv64ima --with-abi=lp64``` to the configure command line when you build riscv-gnu-toolchain.

esolera, 2017-10-03:
I okey thanks, today I´m going to tried this build. If something happend I will post it. Thanks

jhcloos, 2017-10-03:
@aswaterman 
Ah.  Good.  But that means that multilib should be the default.

esolera, 2017-10-03:
@aswaterman  The compilation was well. Now I have all the test in RV64IMA but a build a rocket with this instruction set. The rocket build adding this configuration in config file in system:

class Default64sinFDsinC extends Config( new WithoutFPU ++ new WithoutCompressed ++ new DefaultConfig )

The build run well. When a tried to run my benchmark fir test, I got this error : 

./emulator-freechips.rocketchip.system-Default64sinFDsinC +max-cycles=100000000 +verbose output/fir.riscv 3>&1 1>&2 2>&3 | /home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/bin/spike-dasm  > output/fir.riscv.out && [ $PIPESTATUS -eq 0 ]
Makefile:40: recipe for target 'output/fir.riscv.out' failed
make: *** [output/fir.riscv.out] Error 1

In the next link are all log for building the test, the log for run fir.c  in RV64IMA rocket  and also the link have  the fir.c test. I think maybe is a memory problem with the scratchpad but I'm not sure how I can make it bigger.  This is the link, if you need something else please say me. 

https://www.dropbox.com/sh/cvqezzxmr9x3dq9/AACohIhNah3tRQqWZZiUPVjpa?dl=0

aswaterman, 2017-10-03:
@jhcloos it's really up to the distro.  For example, the default on Ubuntu for x86-64 is to _not_ ship the IA-32 multilibs these days...

aswaterman, 2017-10-03:
An indirect jump at pc=80001fc4 (line 161816 in the log) is going to a bad address.  Unfortunately I don't have time to look into it further, but you can look at the objdump for your program and try to see why things are going wrong.