riscv-software-src/riscv-tests #85: ISA tests for 32-bit double-precision (e.g. `rv32ud`)

tovine, 2017-10-23:
I noticed that there is no ISA test for the 32-bit implementation with hardware double-precision support (e.g `rv32ud`, while both `rv32uf` and `rv64ud` exist - why? RV32D is an extension that is listed in the ISA spec, so it would only be natural that it also should have unit tests?

If there are no currently existing tests or plans to release such tests I will have to make them myself (a few of the rv64ud tests seem to work as is though).

aswaterman, 2017-10-23:
It just hasn't been done yet.  Generalizing the RV64UD tests is the way to go, just like the RV32UF tests (which override some macros then use the RV64UF tests).

tovine, 2017-10-23:
I see. Is it on the near horizon (like this week or the next) or should I just try porting them myself and then submit a pull request?

If I'm to do the porting it would be nice with some pointers as to what needs changing and what to look out for. Is there a "golden" reference implementation that can be used to verify the correctness of the tests (I'm guessing Spike)?  
Wouldn't make sense to use my model for comparison as it's the one that needs verification... :)

aswaterman, 2017-10-23:
It's not on the near horizon, but I'd be happy to review a pull request that adds them, if you follow the same porting strategy as rv32uf.

Yes, Spike's considered the reference.

tovine, 2017-11-07:
I'm looking into the porting now, but I have one question about how to verify the computation results - in the rv64ud tests the result is moved back to the integer registers after computation using `fmv.x.d` instruction but that doesn't exist for rv32.

My suggestion would be to use `fsd` to store the result somewhere in memory (any suggestions?) and then load it word by word into integer registers before comparing...
Am I onto something or is there a simpler way to do it?

aswaterman, 2017-11-07:
Good point, I think fsd + lw + lw is the way to go. You can use some space in the data section for the temporary. You can make new macros that do this instead of fmv, then do some undef/define stuff so the tests are forced to use the new macros.

tovine, 2017-11-08:
I started implementing something here yesterday: https://github.com/tovine/riscv-tests/commit/2ad52c0a098445898f53825910d10451a797cd07

It seems to work for me at least (forgot to commit `rv32ud/fadd.S` but it's basically copied from the similar one in `rv32uf`), will try the same with the others now.

tovine, 2017-11-08:
Almost done porting everything execpt for `structural.S` and `move.S` now.

I haven't tested it against Spike yet (have not gotten around to setting it up) but it seems to behave in a way that makes sense with my simulated model.
Current diff: https://github.com/riscv/riscv-tests/compare/master...tovine:rv32ud_tests

tovine, 2017-11-12:
Finally got around to setting up Spike, all tests ported so far seem to work (everything from `rv64ud` except `move` and `structural`).
I also created a Makefrag for `rv32ud` (commit ad407fb7e62c0214a2af9d916e8b7d339668adb1) and limited the testing to rv32* when running with `XLEN != 64` (in 85b5fbe757436066904f0b6754b94a86dbbbb1fa)

aswaterman, 2017-11-12:
Cool, I can review a pull request once you think it's ready to go.