riscv-software-src/riscv-tests #89: SBREAK test does not check EPC

ccelio, 2017-10-31:
EBREAK and ECALL set EPC register to the address of the EBREAK/ECALL instruction, per the Privileged Specification v1.11. 

The `rv64si/sbreak.S` test only checks the `cause` register, not the `epc` register.

Looks like I've been called out: https://github.com/ucb-bar/riscv-boom/issues/39

ccelio, 2017-10-31:
I think this should do it:

````diff
riscv-tests/isa/rv64si$git diff sbreak.S
diff --git a/isa/rv64si/sbreak.S b/isa/rv64si/sbreak.S
index c2a6e49..99ff62b 100644
--- a/isa/rv64si/sbreak.S
+++ b/isa/rv64si/sbreak.S
@@ -24,6 +24,7 @@ RVTEST_CODE_BEGIN
 
   li TESTNUM, 2
 
+do_break:
   sbreak
   j fail
 
@@ -34,6 +35,9 @@ stvec_handler:
   li t1, CAUSE_BREAKPOINT
   csrr t0, scause
   bne t0, t1, fail
+  la t1, do_break
+  csrr t0, sepc
+  bne t0, t1, fail
   j pass
 
 RVTEST_CODE_END
````

aswaterman, 2017-10-31:
LGTM, can you PR this change?

PR #92 (2017-11-01): SBREAK test now checks EPC value.
Closes #89
commit f2816253 (2017-11-01): SBREAK test now checks EPC value. (#92)

Closes #89
closed 2017-11-01 by aswaterman