riscv-software-src/riscv-tests #93: debug: assert threads failure

mwachs5, 2017-11-01:
I am running riscv-tests against a simulator, and get an assertion failure after GDB attempts to do `info threads`. If I run everything manually, everything works fine, but I am suspicious that it may be attempting to do `info threads` before GDB has finished connecting to OpenOCD. However, with this error, no GDB log is generated so it is hard to debug.

What timeout/handshake is used to ensure that `info threads` is not called before GDB finishes connecting to OpenOCD?

Also the code which printed the temporary GDB file paths to the stdout has been removed, which makes it impossible to monitor the progress of the tests on a simulation.

mwachs5, 2017-11-01:
The issue is a timeout one. The `set remotetimeout` command should be called BEFORE the `target extended remote localhost` command. This is not being done currently. I will have a PR with this change.

https://github.com/riscv/riscv-tests/blob/master/debug/testlib.py#L343

This addresses the thing I was trying to debug, but not the issue that it's very difficult to debug without the GDB output, which is not getting created in this failure mode even though GDB has already started running.

mwachs5, 2017-11-02:
I am closing this as it was fixed by #94

closed 2017-11-02 by mwachs5

timsifive, 2017-11-02:
> with this error, no GDB log is generated so it is hard to debug.

Fixed in d2bf7e8007f2f8972f3974ab985c26cba38c18ec