riscv-software-src/riscv-tests #95: Temporary GDB log files are not actually created

mwachs5, 2017-11-02:
I do not think the temporary GDB log files are actually getting created:

```Temporary logs: simv.log, /tmp/openocdd0LJbx.log, /tmp/gdb@34495-XDE7eE.log, /tmp/gdb@46522-DwDLMA.log
[ComplianceAndPassTest] pass in 434.44s
:::::::::::::::::::::::::::[ ran 1 tests in 434s ]::::::::::::::::::::::::::::
1 tests returned pass

timsifive, 2017-11-02:
They get deleted after the test has passed. Then they should show up in the log file. Apparently that doesn't always happen, according to #93, which is definitely a bug.

mwachs5, 2017-11-02:
They show up only AFTER the `info threads` command executes successfully (at least). Which means if that command fails for some reason, they don't show up at all.

timsifive, 2017-11-02:
The log files are created before gdb is started. Their names are not printed until later (so they can all be printed together, reducing output clutter). Is that a problem that needs fixing?

mwachs5, 2017-11-02:
it is a problem if GDB starts, fails, and there is no record of the matter. Try changing your "assert threads" to "assert false" and you will see what I mean. Or are you saying that even in that case, there would be a GDB report in the "full" logfile? That was not my experience.

mwachs5, 2017-11-02:
To be clear, it can take several minutes for GDB to execute the "connect" (that you added in your fix) to a simulation target. So, I wouldn't say that it's useless to print the log file as soon as possible. It's not a deal breaker though.

timsifive, 2017-11-02:
I agree that that was a bug. I just fixed it in d2bf7e8007f2f8972f3974ab985c26cba38c18ec.
Now, even in that case, there will be a GDB report in the full logfile.

How do you feel about a command line option that causes logfile names to be printed ASAP?

mwachs5, 2017-11-02:
A command line option is better than nothing, but is it really that horrible to print it out? You print it out anyway, just later. Anyway I don't feel strongly enough to argue against the command line option.

timsifive, 2017-11-02:
The thing is if you print it out ASAP then you end up with multiple lines.
(Or you need to add some logic to predict which log files will be required,
and create them all up front.) This is all for information that most people
don’t care about at all.

For me, my concern is to keep the output of make all (running all tests
against 4 different spike targets simultaneously) readable.
​

timsifive, 2017-11-02:
Forgot to mention this issue in change 6f7ebb610d6bb8817a9592cc06a7d108381f1761

closed 2017-11-03 by mwachs5