riscv-software-src/riscv-tests #99: fmin sNaN propagation

LeeMooreImperas, 2017-11-10:
Hi
I have noted that the behavior of the sNaN propagation as per the test for rv64uf-p-fmin has changed since the release of v2.2 of the ISA Specification
In the 2.2 spec an sNaN on any input of FMIN/FMAX should generate a canonical NaN
This is no longer the expectaion with either the current test or spike.

In order to have a set of tests and a version of spike which reflects the specification as per 2.2, how is this to be done ?
as far as I can see there are no 'releases' bound to the version of the ISA, I would like to have a set of tests and golden reference simulator which matches the 2.2 spec.

Thx
Lee

aswaterman, 2017-11-10:
Sorry for not versioning the releases; we should start doing that.  Indeed
Spike and the tests correspond to v2.3-draft, which defines FMIN to behave
like the IEEE 754 function minimumNumber.  (You can find the up-to-date
spec here: https://github.com/riscv/riscv-isa-manual)

LeeMooreImperas, 2017-11-13:
Hi Andrew,
Thanks for the explaination.
Is it possible for me to do a checkout of spike and riscv-tests, on a particular date, in which the date will correspond to the ISA spec as per ver 2.2 
Thx
Lee

aswaterman, 2017-11-13:
If you look at the git log for the file riscv/insns/fmin_s.h in Spike, you should be able to find the commit where this behavior changed. The date of that commit will probably match the date that the test program changed, too.