riscv-software-src/riscv-tests #101: Running Virtual Memory tests on Spike

LeeMooreImperas, 2017-11-16:
Hi All,

I am trying to run the -v- tests on spike, but I am getting unexpected behavior, so for example

spike -l --pc=0x80000000 --isa=RV32IMAFD riscv-tests/isa/rv32ui-v-addi 

this does not seem to do what I expect.
At the point where the sret is called  I get the following
```
core   0: 0xffffffff800000c0 (0x10200073) sret
core   0: exception trap_instruction_access_fault, epc 0x00000000000029e8
core   0:           badaddr 0x00000000000029e8
core   0: 0x00000000ffc000c4 (0x00000000) addi    s0, sp, 0
core   0: exception trap_illegal_instruction, epc 0x00000000ffc000c4
core   0:           badaddr 0x0000000000000000
core   0: 0x0000000080000008 (0x2600206f) j       pc + 0x2260
```

this then ends up in a loop, and eventrually ends with
```
core   0: 0xffffffff8000224c (0x00f76733) or      a4, a4, a5
3 0xffffffff8000224c (0x00f76733) x14 0x0000000000000000
core   0: 0xffffffff80002250 (0xfe070ae3) beqz    a4, pc - 12
3 0xffffffff80002250 (0xfe070ae3)
*** FAILED *** (tohost = 420)
```

I presume I am configuring spike incorrectly, as the tests are built simpy by configure/make

Is there a procedure I can follow which describes
1. The building of the conformance tests
2. Execution of the tests on spike to validate the tests operate correctly

Many Thx
Lee

eroom1966, 2017-11-22:
Closing this issue.
Seems the problem was an inconsistency between releases of spike and the tests
Now working correctly having checked out the tests and spike for the same tag

Thx
Lee

eroom1966, 2017-11-22:
resolved #101

LeeMooreImperas, 2017-11-22:
Closing this issue.
Seems the problem was an inconsistency between releases of spike and the tests
Now working correctly having checked out the tests and spike for the same tag

Thx
Lee

closed 2017-11-22 by LeeMooreImperas