riscv-software-src/riscv-tests #112: Error information *** FAILED *** (tohost = 1337)

pkulining, 2018-01-08:
Hi. 

I'm trying to add indexed load and store instruction into riscv, and I changed the simulator and gnu-toolchain and built successfully.But when I tried to run the .riscv file in riscv-test/build/benchmark,I got the error information " *** FAILED *** (tohost = 1337) " . I would like to know  in what case that I will get such error information ,for example,loading data from illegal address? 

And I find "tohost" is set to 1337 in  handle_trap function, then I find trap_entry in  crt.S ,but I can't find which file calls or jumps to the trap_entry. So I want to know the mechanism of spike command to discover the trap and handle it.

Thanks!

noureddine-as, 2018-05-07:
Hi,

Exceptions aren't handled there, so the program just stops by sending 1337 to the front end server via HTIF interface. You can examine the cause of the exception that ends the program by seing mcause CSR.

Best regards.

omerguzelelectronicguy, 2022-05-18:
Hi, I have the same case and I didn't get one point. Does the program stopped by the spike? or stopped by the executable itself by sending an infinite loop?

omerguzelelectronicguy, 2022-05-23:
By the way, I solved my problem by giving the memory adress as input to the spike like this: `spike -d --log-commits -m0x40000000:0x81920,0x20000000:0x100 uart_test.riscv`.