riscv-software-src/riscv-tests #124: Bug in isa/rv64mi/illegal.S?

diegonehab, 2018-04-16:
One of the tests is failing on an emulator I am working on.

      # Test vectored interrupts if they are supported.
    test_vectored_interrupts:
      csrwi mip, MIP_SSIP
      csrwi mie, MIP_SSIP
      la t0, mtvec_handler + 1
      csrrw s0, mtvec, t0
      csrr t0, mtvec
      andi t0, t0, 1
      beqz t0, msip
      csrsi mstatus, MSTATUS_MIE
    1:
      j 1b
    msip:
      csrw mtvec, s0

      # Delegate supervisor software interrupts so WFI won't stall.
      csrwi mideleg, MIP_SSIP
      # Enter supervisor mode.
      la t0, 1f
      csrw mepc, t0
      li t0, MSTATUS_MPP
      csrc mstatus, t0
      li t1, (MSTATUS_MPP & -MSTATUS_MPP) * PRV_S
      csrs mstatus, t1
      mret

    1:
      # Make sure WFI doesn't trap when TW=0.
      wfi
    bad3:
      .word 0
      j fail

In the snippet above, right before mret, we have SSIP and SSIE set and we are delegating SSIP. After mret, we are in S-mode. So even before WFI is executed, the processor can trap to stvec, right? However, stvec has never been set. I don't think that was intended. Is this a bug in the test itself?

aswaterman, 2018-04-16:
Supervisor interrupts are globally disabled at this point (sstatus.SIE=0), so even though SSIP and SSIE are set, there should be no interrupt taken.

diegonehab, 2018-04-17:
Ah, thanks!

closed 2018-04-17 by diegonehab