riscv-software-src/riscv-tests #134: rv64ua/lrsc.S reads from un-initialized memory.

ccelio, 2018-04-30:
**Background:**

The `lrsc` test performs a `load-reserve` from un-initialized memory. The write back value is `dontCare`, however, this can be problematic for co-simulation.

https://github.com/riscv/riscv-tests/blob/835f252ebe59ef782e96f467462a03754ea87936/isa/rv64ua/lrsc.S#L34-L41

It loads the address of `foo`, but then adds 1024 to it, in the hopes that 1024 is beyond the reach of the reservation (a painfully implementation-dependent value).

--------
**Question:**

Is leaving `foo+1024` un-initialized an oversight, or a deliberate choice? Is it wanting a smaller static binary to hasten loading?

aswaterman, 2018-04-30:
The latter, but expanding the data section seems OK.

This test obviously needs to be rewritten at some point.

ccelio, 2018-04-30:
Alright, I'm taking this on now. Question:

How do I provide some fixed spacing between two labels (e.g., between `foo` and `foo2`?) Is that `.align x`? That seems to enforce alignment, but doesn't work for the case where foo+4 hits the desired alignment. Also, what is x? `Size` or `log(size)`?

ccelio, 2018-04-30:
FYI, the change in test time is not noticeable (the whole thing takes 9.7 seconds for my setup, +/- 0.3 s).

aswaterman, 2018-04-30:
```.skip 1024```

PR #135 (2018-05-01): [rv64ua/lrsc] Initialize memory read out.
Even though the load contents are discarded, this un-initialized memory value
can lead to a divergence for co-simulation between two different RISC-V designs.

Closes #134.
closed 2018-05-01 by aswaterman