riscv-software-src/riscv-tests #136: Can I run the rv32 tests on a rv64 architecture?

gameboo, 2018-05-04:
I was trying to test a rv64 implementation (happy running rv64 tests) with rv32 tests and expected to maybe find bugs in the style of "you didn't sign extend things correctly" or the likes.

Quite a few tests were happy, and, sure thing, some other tests failed. However, looking a bit more into it, it seems like the `TEST_CASE` macro cannot cope with a test built for rv32 when ran on rv64:
```asm
#define TEST_CASE( testnum, testreg, correctval, code... ) \
test_ ## testnum: \
    code; \
    li  x29, MASK_XLEN(correctval); \
    li  TESTNUM, testnum; \
    bne testreg, x29, fail;
```
If I understand correctly, `li` (and ultimately `lui`) will sign extend the value passed to `MASK_XLEN` according to the specified behaviour when running in 64 bit mode. In particular, that looks to me like it breaks things like `TEST_IMM_OP( 12, addi, 0x000000007ffff7ff, 0x000000007fffffff, 0x800 );` (used in [isa/rv64ui/addi.S](https://github.com/riscv/riscv-tests/blob/master/isa/rv64ui/addi.S) linked from [isa/rv32ui/addi.S](https://github.com/riscv/riscv-tests/blob/master/isa/rv32ui/addi.S)) (and I do fail the test in this way).

Just as a quick sanity check, I did something like this:
```asm
#define TEST_CASE( testnum, testreg, correctval, code... ) \
    code; \
    li  x29, MASK_XLEN(correctval); \
    li  x28, 1; \
    sll x28, x28, (__riscv_xlen - 1); \
    sll x28, x28, 1; \
    addi x28, x28, -1; \
    and x29, x29, x28; \
    and testreg, testreg, x28; \
    li  TESTNUM, testnum; \
    bne testreg, x29, fail;
```
and the test (along with a few others that were previously failing) started passing. Some other tests still fail for things that I suspect are similar reasons (and possibly some bugs in the rv64 implementation itself).

Are rv32 tests simpl not expected to be run on rv64 implementations? Or am I missing a step somewhere maybe ?

aswaterman, 2018-05-04:
RV32 tests aren't intended to run in RV64.

gameboo, 2018-05-04:
fair enough :)

aswaterman, 2018-05-04:
I would say it's a feature that the tests don't run correctly on RV64 systems.  Put a different way, if all RV32 tests ran correctly on native RV64, that would mean their coverage was poor: they can't tell RV32 and RV64 apart!

The canonical way to solve this problem is to implement MXL and switch to RV32 mode.

gameboo, 2018-05-04:
Yes. I edited my message a bit late maybe, but the MXL and mode switch at the beginning of the test is what I meant. I was thinking you'd want to be able to test that your rv64 implementation can correctly run code that is supposed to run on an rv32 implementation, and maybe explicitly test separately the behaviour that currently cause rv32 to fail on rv64 for explicit coverage... Not worth the hassle?