riscv-software-src/riscv-tests #137: What are the meaning of "tohost" and ".tohost"?

clx7913, 2018-05-06:
I've looked up the 
"riscv_test.h"  (in https://github.com/riscv/riscv-test-env/blob/68cad7baf3ed0a4553fffd14726d24519ee1296a/p/riscv_test.h)
and
"test_macros.h" (in https://github.com/riscv/riscv-tests/blob/master/isa/macros/scalar/test_macros.h),but still wonder that what's the meaning of "tohost" and ".tohost"?

konkitoman, 2025-02-18:
From what I understand `tohost` is used by the test runner to read the result of the tests.

If `tohost & 1 == 1` the result was written, the result is `tohost >> 1` if is zero all test passed else is the test number that failed.

pdonahue-ventana, 2025-02-18:
See https://github.com/riscv-software-src/riscv-isa-sim/issues/364#issuecomment-607657754