riscv-software-src/riscv-tests #138: How to run dhrystone test on Pulpino Core

ellenFan, 2018-05-11:
Hi all

I tried to run the dhyrstone test on Pulpino RISCY Core by adding source code (dhrystone.c  dhrystone.h  dhrystone_main.c  encoding.h /*from env folder*/  util.h) and used cmake to add this application (details on how to add an application to pulpino could be found in the following link:https://github.com/pulp-platform/pulpino/blob/master/sw/README.md ) 

It compiled but when I run the pulpino-converted spi version of the dhrystone application on the zedboard, I got the following error:
PULPino: Measured time too small to obtain meaningful results
I also checked the start time and end time, they are all zero. I tried to increase the number of runs but even if it takes longer time(longer than one second) to give a feedback, start/end time are also zero.

According to dhrystone.h, the Start_Timer and End_Timer functions in my case are defined as:
#define Start_Timer() Begin_Time = read_csr(mcycle)
#define Stop_Timer() End_Time = read_csr(mcycle)

Someone else online also has similar issues when testing dhrystone on Cortex-M4 and their problem is caused by missing time measurement functions in the testing environment. They finally solved it by using Cortex-M4's own system timer to count cycles. (Here's the link 
http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.faqs/ka16826.html)

What might be the causes in my case? If time measurement functions are missing, what other source code or path should I add? Our system is Ubuntu 16.04.4, the riscv tool chain of our system is provided by pulpino (link: https://github.com/pulp-platform/ri5cy_gnu_toolchain)

Really appreciate your help

Irdroid, 2023-01-11:
Same issue here on Pulpissimo, did you manage to find a fix for that?

Irdroid, 2023-01-17:
We have managed to compile and run the Dhrystone CPU benchmark tests on Pulpissimo, [you can have a look at the code in our GitHub repo.](https://github.com/Irdroid/Pulp_Dhrystone)

The reason you get zero counting for the cycles counter and instructions counter is because Pulpino/Pulpissimo RISC-V CPUs have non-standard RISC-V cycles/instructions counter registers, thus modification to the code is needed.