riscv-software-src/riscv-tests #144: How to run the benchmarks using proxy kernel in Rocket Chip emulator or FPGA frontend server?

asmitde, 2018-05-31:
Hello,

How do I run the compiled benchmarks through the proxy kernel on the RocketChip emulator?
Eg, I tried running the qsort.riscv in the emulator and it gives me the following error:
[image]

I get the same issue when running on the FPGA:
[image]

The same program when run on spike without the proxy kernel, works fine.
[image]

What can be the issue?

aswaterman, 2018-05-31:
Don't use the proxy kernel.  You should be able to run the tests directly on the RTL simulator or the Zynq frontend server, without pk.

asmitde, 2018-05-31:
Thanks a lot! It worked without the pk.

I have a couple of questions, though.
1) What is the difference between mcycle and minstret?
2) When should we be using pk to run the riscv-elf binaries and when not? (Any programs I compiled manually with the riscv-gnu-toolchain needed pk to run, whereas the benchmarks don't need it, so I'm a bit confused how that works.)

Thanks for your help.

muhammadali201, 2018-09-30:
@asmitde 

Did you get answers to your questions?

colinschmidt, 2018-09-30:
`mcycle` is the number of cycles
`minstret` is the number of instructions
These can differ because of cache misses, branch mispredicts, pipeline flushes, etc.

The benchmarks have a bare-metal library/runtime that lets them run without pk.

Normal programs will need pk, or more (i.e. linux), depending on how much of a system them expect. pk doesn't implement a full kernel.

closed 2018-09-30 by colinschmidt

AbhijittDhavlle, 2020-04-29:
How do you compile a C program for running on the fesvr-zynq or the RISCV Linux on rocket chip? I compiled initially with riscv62-unknown-elf-gcc and it executes with "spike pk hello". But the same compiled binary does not execute on the fesvr(with or without pk). How do I ensure that the compiled binary is RISCV compatible? Thanks ! Please reply asap!