riscv-software-src/riscv-tests #146: std::runtime_error what(): bad syscall

IC-Dream, 2018-06-28:
Hi, 

I modified the "RISCV_GCC_OPT" in makefile of benchmarks like below:

Before modification:
RISCV_GCC_OPTS ?= -DPREALLOCATE=1 -mcmodel=medany -static -std=gnu99 -O2 -ffast-math -fno-common -fno-builtin-printf

After modification:
RISCV_GCC_OPTS ?= -DPREALLOCATE=1 -mcmodel=medany -static -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf

Only changed the "-O2" to "-O3", and then run the program on spike, got the following error:
Commmd: spike dhrystone.riscv
Error: terminate called after throwing an instance of 'std::runtime_error'
  what():  bad syscall #2147507328

@jim-wilson , what can be the issue? 

Thanks for your help

jim-wilson, 2018-06-28:
I managed to reproduce using riscv-tools.  Using the spike -l option, I see that in the _init function, when it calls memset to clear the tbss section, memset goes into an infinite loop.

Looking at memset, I see that when you compile with -O3, gcc recognizes that the code is memset, and substitutes a call to memset, which is normally faster than a user written memset as the library is expected to have hand tuned memset code.  Except that since this is the library itself, making memset call itself doesn't work.

memset is a reserved function name, so it isn't valid for user code to define it.  It is OK for library code to define it, but library code must be careful to avoid this gcc optimization.  Newlib uses -fno-builtins for this purpose.  There are also other ways to solve this problem.

IC-Dream, 2018-06-29:
@jim-wilson,Thank you very much for your explanation. I added the -fno-builtin and the error disappered, but the score of dhrystone benchmark got worse. what's going on? 

Thanks for your help

aswaterman, 2018-06-29:
It probably prevented dhrystone from inlining strcpy. Try
-fno-builtin-memcpy instead?

jim-wilson, 2018-06-29:
-O3 does not necessarily give better performance than -O2.  -O3 enables optimizations that trade space for speed, and sometimes this gives faster code and sometimes it doesn't.  The FSF recommends that you use -O2.  -O3 is there for people who want to try it, and use it if it gives faster working code.

IC-Dream, 2018-06-29:
@aswaterman, it didn't work using -fno-builtin-memcpy, and would introduce the "std::runtime_error what(): bad syscall". 

Thanks for your help

IC-Dream, 2018-06-29:
@jim-wilson, Thanks for your help, i will try it.

aswaterman, 2018-06-29:
Sorry, as Jim mentioned, I meant "-fno-builtin-memset", not "-fno-builtin-memcpy".

IC-Dream, 2018-06-29:
@jim-wilson @aswaterman it still reported the same error after disable all three：memcpy,memset,memmove. only "-fno-builtin" works.

closed 2018-07-02 by IC-Dream

manasabatch4, 2022-03-03:
Hi I am getting bed syscall  error  when I am running in spike I just copy the make file from one  folder to another folder i did change any thing in make file