riscv-software-src/riscv-tests #149: ISA tests are compiled with RVG, instead of the compiler's default march flag.

ccelio, 2018-07-26:
https://github.com/riscv/riscv-tests/blob/73a4ba0b3a218f31b2e1db9c438a13c951d640a3/isa/Makefile#L74-L91

Is there a reason RVG is hard-coded for the compilation of the ISA tests, instead of RVGC as specified by the compiler's default march flag? 

For RVC cores, this would seem to reduce coverage unnecessarily. Or is there a fundamental aspect of the riscv-tests-envs that need aligned instructions?

colinschmidt, 2018-07-27:
Perhaps you can't specify rv32 without specifying the full ISA string?

ccelio, 2018-07-31:
Let me clarify: it won't compile the ISA tests with rv64gc even though the gcc compiler is rv64gc by default. I'm wondering if this is intended behavior.

jerryz123, 2019-01-04:
Has anything changed with respect to this? It would be nice to have more comprehensive RVC coverage.
The benchmarks also default to rv64g, at least with my version of riscv-gnu-toolchain.

aswaterman, 2019-01-04:
Yes, we had to add some alignment directives to the virtual-memory environment so it would work with RVC, but it all works now.  (You're still on your own for the Makefile changes.)

closed 2019-01-04 by aswaterman