riscv-software-src/riscv-tests #155: PrivChange test is disabled

mwachs5, 2018-08-27:
I think the problem with this test is that it doesn't clear the `satp` register and other memory translation related registers before setting priv to U-mode. Then it relies on getting an exception so that the PC doesn't advance normally once the core is sent to U-mode.

But consider the scenario where some other test or process has set up the system so that valid user instructions are placed in the 0x8000_0000 range, and set up virtual memory so that the system executes user code out of that range. Then, *even if the system is reset*, those registers remain set up (very little is specified to be reset in the RISC-V ISA). So, when the core is asked to go into U-mode, it's fine with it because everything is all set up for it (normally, the boot SW would clear these registers to 0 or otherwise set them up properly before executing an `mret` or `sret`).

OpenOCD sees the PC advancing, but doesn't do any address translation so it thinks that User mode must somehow be executing the M-mode only CSR instructions. But actually the PC may be getting translated and pointing to something that is valid for U-mode.

Thoughts? What am I missing here?

timsifive, 2018-08-27:
Instead of forcing each test to reset the parts of the system it might be sensitive to, it makes more sense to require that the OpenOCD configuration script performs this reset. In theory `reset halt` at the end of the script takes care of that. Does that fix the problem?

mwachs5, 2018-08-27:
If what I am saying is true, no, having a `reset halt` at the end of the OpenOCD script would not solve the problem. Because so little is required to actually be reset by the RISC-V spec. 

Changing the privilege mode is like doing an "mret" without performing whatever setup needs to be done before doing it. Usually a system and the SW running on it will set whatever needs to be set before executing such an instruction.

mwachs5, 2018-10-16:
Fixed by #156

closed 2018-10-16 by mwachs5