riscv-software-src/riscv-tests #162: question about load data when run spike

IC-Dream, 2018-09-12:
Hi,

There is a self-checking directed test case in ~/riscv_tests/isa/rv32ui/lb.S to verify the load instruction. 

For example:
Assembly:
[image]

Disassembly:
[image]

Commit log
[image]

The load would load data back correctly and wrote to x30 correctly, but i don't know how the spike know the load data=32'hffff_ffff of the address=32'h80002000 ? Does the lb.S initialize this address to 32'hffff_ffff ? or something else ?

I am struggling to this, any help would be appreciated. 

Thanks
Dream

IC-Dream, 2018-09-12:
More info: we didn't do any write to this address before the load.

IC-Dream, 2018-09-13:
@jerralph here are the related code, but why the data of 0x80002001 ~ 0x80002003 are not 0x00/0xf0/0x0f respectively? 
Assembly:
[image]
Disassembly:
[image]

Thanks

aswaterman, 2018-09-13:
`lb` loads only the byte at `80002000`, then sign-extends it to the register width.  So the bytes `80002001`-`80002003` do not matter.

IC-Dream, 2018-09-13:
but the test 3/4/5 need to load data from the address 0x80002001-0x80002003, right ?
[image]

Thanks

aswaterman, 2018-09-13:
Yeah.  Test 3 loads the byte at 80002001 (which has value 00), then sign-extends the byte, giving the result 00000000.

Test 4 loads the byte at 80002002 (which has value f0), then sign-extends the byte, giving the result fffffff0.

etc.

aswaterman, 2018-09-13:
Is your question about the disassembly?  The disassembler is trying to treat these bytes as instructions, which of course makes no sense.  If you disassemble with different arguments (say, -d instead of -D), maybe you'll see the bytes as you expect.

aswaterman, 2018-09-13:
I see. It’s not a riscv-tests question anymore, then. You can try other objdump flags, or use xxd or something.

closed 2018-09-13 by aswaterman