riscv-software-src/riscv-tests #168: How to compile the test code dot S

mongsim, 2018-10-22:
Can anyone show an example of how to compile the tests code .s files with riscv64-unknown-elf-gcc.

tovine, 2018-10-23:
Could you be a bit more specific about which .S files?
Usually one just compiles and runs a specific test by navigating to the directory with the `Makefile` (like this one: https://github.com/riscv/riscv-tests/blob/master/isa/Makefile) and runs `make` from there...

mongsim, 2018-11-05:
Hello Torbjorn,

Thank you for your fast reply!

I tried this simple command line compilation and got these errors. Any help is greatly appreciated.

riscv64-unknown-elf-gcc -c -mabi=ilp32 -march=rv32im -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles add.s -o add.o

add.s: Assembler messages:
add.s:13: Error: unrecognized opcode `rvtest_rv32u'
add.s:14: Error: unrecognized opcode `rvtest_code_begin'
add.s:20: Error: unrecognized opcode `test_rr_op( 2,add,0x00000000,0x00000000,0x00000000)'
add.s:21: Error: unrecognized opcode `test_rr_op( 3,add,0x00000002,0x00000001,0x00000001)'
add.s:22: Error: unrecognized opcode `test_rr_op( 4,add,0x0000000a,0x00000003,0x00000007)'
add.s:24: Error: unrecognized opcode `test_rr_op( 5,add,0xffff8000,0x00000000,0xffff8000)'
add.s:25: Error: unrecognized opcode `test_rr_op( 6,add,0x80000000,0x80000000,0x00000000)'
add.s:26: Error: unrecognized opcode `test_rr_op( 7,add,0x7fff8000,0x80000000,0xffff8000)'
add.s:28: Error: unrecognized opcode `test_rr_op( 8,add,0x00007fff,0x00000000,0x00007fff)'
add.s:29: Error: unrecognized opcode `test_rr_op( 9,add,0x7fffffff,0x7fffffff,0x00000000)'
add.s:30: Error: unrecognized opcode `test_rr_op( 10,add,0x80007ffe,0x7fffffff,0x00007fff)'

Thank you,

Mong

#ifndef _ENV_PICORV32_TEST_H
#define _ENV_PICORV32_TEST_H

#ifndef TEST_FUNC_NAME
#  define TEST_FUNC_NAME mytest
#  define TEST_FUNC_TXT "mytest"
#  define TEST_FUNC_RET mytest_ret
#endif

#define RVTEST_RV32U
#define TESTNUM x28

#define RVTEST_CODE_BEGIN		\
	.text;				\
	.global TEST_FUNC_NAME;		\
	.global TEST_FUNC_RET;		\
TEST_FUNC_NAME:				\
	lui	a0,%hi(.test_name);	\
	addi	a0,a0,%lo(.test_name);	\
	lui	a2,0x10000000>>12;	\
.prname_next:				\
	lb	a1,0(a0);		\
	beq	a1,zero,.prname_done;	\
	sw	a1,0(a2);		\
	addi	a0,a0,1;		\
	jal	zero,.prname_next;	\
.test_name:				\
	.ascii TEST_FUNC_TXT;		\
	.byte 0x00;			\
	.balign 4, 0;			\
.prname_done:				\
	addi	a1,zero,'.';		\
	sw	a1,0(a2);		\
	sw	a1,0(a2);

#define RVTEST_PASS			\
	lui	a0,0x10000000>>12;	\
	addi	a1,zero,'O';		\
	addi	a2,zero,'K';		\
	addi	a3,zero,'\n';		\
	sw	a1,0(a0);		\
	sw	a2,0(a0);		\
	sw	a3,0(a0);		\
	jal	zero,TEST_FUNC_RET;

[... 576 lines cut ...]
  bne a0, a3, fail; \
  j 1f; \
  .align 3; \
  test_ ## testnum ## _data: \
  .double result; \
1:

#-----------------------------------------------------------------------
# Pass and fail code (assumes test num is in TESTNUM)
#-----------------------------------------------------------------------

#define TEST_PASSFAIL \
        bne x0, TESTNUM, pass; \
fail: \
        RVTEST_FAIL; \
pass: \
        RVTEST_PASS \

#-----------------------------------------------------------------------
# Test data section
#-----------------------------------------------------------------------

#define TEST_DATA

#endif

mongsim, 2018-11-08:
> That looks a bit strange, when I run make (from the riscv-tests/isa dir) this is what comes out:
>
> riscv32-unknown-elf-gcc -march=rv32g -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -I./../env/p -I./macros/scalar -T./../env/p/link.ld rv32mi/shamt.S -o rv32mi-p-shamt
> riscv32-unknown-elf-objdump --disassemble-all --disassemble-zeroes --section=.text --section=.text.startup --section=.text.init --section=.data rv32mi-p-shamt > rv32mi-p-shamt.dump
>
> It seems like what you're missing is I./../env/p -I./macros/scalar -T./../env/p/link.ld (update relative paths to match your working directory) because the test_rr_op is a preprocessor macro defined in riscv-tests/isa/macros/scalar/test_macros.h

Thank you Torbjorn, I got it working.

tovine, 2018-11-08:
Glad I could help! 😄

mongsim, 2018-11-22:
Hello Torbjorn,

I have two questions hope you can help 😊 and Happy Thanksgiving!

  1.  Do you have a coremark port that I can use? The trend is towards coremark.
  2.  I an using a 32-bit data memory and would like to take advantage of the data bus width and also to implement later a icache too. Do you have a good solution to handle the 16bit compression and the regular 32-bit instructions?

Thank you,

Mong

tovine, 2018-11-23:
> 1. Do you have a coremark port that I can use? The trend is towards coremark.

Unfortunately I don't, but it's pretty straightforward to port it - you just have to implement the relevant functions that are missing in this file: https://github.com/eembc/coremark/blob/master/barebones/core_portme.c

> 2. I an using a 32-bit data memory and would like to take advantage of the data bus width and also to implement later a icache too. Do you have a good solution to handle the 16bit compression and the regular 32-bit instructions?

I'm sorry, no, I don't have enough experience with that yet.  
However, you can take a look at the RISC-V HW Dev mailing list archives here to see if you can find something useful, or send a question to the list if it hasn't been discussed before: https://groups.google.com/a/groups.riscv.org/forum/#!forum/hw-dev

Good luck with your project!

mongsim, 2019-03-01:
Hello Torbjon,

I was able to continue and is successful with both Dhrystones and Coremark without the compressed instruction or the “c” compiler option.
When using the compile option, the processor passed the Test Suite except two test cases, Jal and Jalr. Is there anything special about this two instructions that is difference from the 32-bit version beside having PC + 2 instead of PC + 4.

Is there a need to modify these two tests (I believe many people have used this test suite) ?

Any help is greatly appreciated 😊

Thank you,

Mong

  1.  I an using a 32-bit data memory and would like to take advantage of the data bus width and also to implement later a icache too. Do you have a good solution to handle the 16bit compression and the regular 32-bit instructions?

tovine, 2019-03-04:
I don't know, sorry.
Haven't really worked enough with the C extension to be familiar with it, then only thing I can recommend is to read the relevant part of the [specifications](https://riscv.org/specifications/) (Section 12.4)

mongsim, 2019-03-04:
Hello Torbjorn,

Thank you for your reply.

I am looking for you git site, l might have lost it. Can send me your git site address.

Thank you,

Mong