riscv-software-src/riscv-tests #185: errors in `build_project riscv-tests --prefix=$RISCV/riscv64-unknown-elf `

Lustrew, 2019-03-22:
There is something wrong when I run `./build.sh`
Here is the detail, could someone tell me how to solve this problem? Thank you!
```shell
/home/lustre/riscv-tools/bin/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld:/home/lustre/riscv-tools/riscv-tests/build/../isa/../env/v/link.ld:2: syntax error
collect2: error: ld returned 1 exit status
make[1]: *** [rv32ui-v-simple] Error 1
make: *** [isa] Error 2
```

mukesh-cst, 2019-03-28:
Is there any fix/update on this?
I'm also facing this error:
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/8.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /tmp/ccOHZueX.o: in function `.L58':
vm.c:(.text+0x248): undefined reference to `__riscv_flush_icache'
collect2: error: ld returned 1 exit status
/M/Installs/riscv/riscv-tests/isa/Makefile:74: recipe for target 'rv32ui-v-simple' failed

aswaterman, 2019-03-28:
These tests aren’t meant to be built with a Linux toolchain. You need to use an embedded (elf) toolchain. Then, the compiler will emit inline code instead of calling __riscv_flush_icache.

wbjsgtc, 2019-09-09:
Any fix? I also have this issue.

closed 2019-10-21 by Lucas-Wye