riscv-software-src/riscv-tests #189: bss section does not be cleared in benchmarks

lizhm82, 2019-04-25:
In benchmark programs, I have noticed that .bss/.sbss section does not be cleared,  and .bss section is not empty. The initialization only clear the .tbss section.

aswaterman, 2019-04-25:
The existing test environments assume the loader will clear bss (as is the case in a managed environment). If your environment does not clear bss, you can use a different test environment or modify the existing ones.

lizhm82, 2019-04-26:
I use the openocd + gdb to run the benchmark program, it didn't clear bss automatically, so I have to modify the code to clear bss, now it works.  
I confused why don't consider adding the bss clear in the code, make it more general?

abhinavfury13, 2019-09-05:
hey @lizhm82  i am using openocd + gdb to test as well and would like to know how you cleared the bss in code? if possible could you please post the code here. Did you make changes to the crt.S file?