riscv-software-src/riscv-tests #190: error: invalid suffix "x" on integer constant

amineejaam, 2019-05-05:
There is something wrong when I run make to build it
Here is the detail, could someone tell me how to solve this problem? Thank you!
``` shell
/Users/cp0/riscv-tests/isa/../env/v/vm.c: In function 'coherence_torture':
<command-line>: error: invalid suffix "x" on integer constant
/Users/cp0/riscv-tests/isa/../env/v/vm.c:197:25: note: in expansion of macro 'ENTROPY'
   unsigned int random = ENTROPY;
/Users/cp0/riscv-tests/isa/../env/v/vm.c: In function 'vm_boot':
<command-line>: error: invalid suffix "x" on integer constant
/Users/cp0/riscv-tests/isa/../env/v/vm.c:212:25: note: in expansion of macro 'ENTROPY'
   unsigned int random = ENTROPY;
make[1]: *** [rv32ui-v-simple] Error 1
make: *** [isa] Error 2
```

CwbhX, 2019-06-10:
I have the same issue trying to build on Mac OS 10.14.5 :/
Can't immediately see why this is occuring

aswaterman, 2019-06-11:
The culprit is this line: https://github.com/riscv/riscv-tests/blob/master/isa/Makefile#L61

It works fine on bash on Ubuntu. If you can figure out how to fix it for your platform without breaking it on this one, please make a PR.

CwbhX, 2019-06-13:
@aswaterman I'm pretty busy at the moment, but thanks for pointing out the root cause. I'll see if I can get it working under Mac OS and then test it in a VM when I get a moment.

aswaterman, 2019-06-13:
👍

dylanmc, 2019-09-04:
I had this error just now (macOS 10.14), and thanks to this issue learned that the build is failing because macOS no longer has `md5sum` installed by default.

```
brew install md5sha1sum
```

fixed my build on this platform.