riscv-software-src/riscv-tests #207: Couldn't open ELF program error

luoxinjie1146, 2019-09-22:
When I run `spike pk rv64ui-p-add`, it reported
```
bbl loader
couldn't open ELF program: vvadd.riscv!
```
I can't make sense how to use the riscv-test.

aswaterman, 2019-09-23:
Don't use pk, just run them directly on spike.

luoxinjie1146, 2019-09-24:
@aswaterman Thanks, it works. But how can I run it on my host? It didn't print PASS or FAIL. Otherwise, what's the difference between `spike` and `pk`?

aswaterman, 2019-09-24:
Spike will exit with a nonzero exit code if the test program fails.

pk is used for running applications compiled for the Linux ABI.

SaltedFishLZ, 2020-12-12:
I wonder how the non-pk version executes some functions like "print"? Will it use HTIF so emulated JTAG in Rocket Chip to send data to the host?

aswaterman, 2020-12-14:
That's right - which is that same thing that pk would do.

SaltedFishLZ, 2021-08-03:
> That's right - which is that same thing that pk would do.

Update: It seems the emulated JTAG isn't the matter here. It seems the only thing we need to load ELF and print messages is the `tohost` and `fromhost` registers. Is my understanding right?
I also wonder how do different emulators recognize these 2 special memory addresses. I searched in the source codes of Rocket Chips but cannot find any clue about how the Rocket emulator achieves this goal.
Moreover, are these 2 addresses hard-coded? If not, how can we change them?
Thanks a lot!

aswaterman, 2021-08-03:
The host system knows about the ELF that's running on the target, and so it knows the addresses from the ELF symbol table.  (The addresses are _not_ hardcoded.)  Then, as long as the host has some mechanism to read the target machine's memory, it all works out.  For the rocket-chip emulator, the host uses the debug interface to read target memory.

SaltedFishLZ, 2021-08-03:
> The host system knows about the ELF that's running on the target, and so it knows the addresses of tohost/fromhost by reading the ELF symbol table. (The addresses are _not_ hardcoded.) Then, as long as the host has some mechanism to read the target machine's memory, it all works out. For the rocket-chip emulator, the host uses the debug interface to read target memory.

So, it means that different emulators may have different implementations to recognize these 2 registers? 

Is there any specification on how to load ELF into Verilator generated emulators? Verilated designs are hard to understand and I cannot find any documentation on modifying the internal states of Verilated designs (e.g., directly modifying the content of the emulated AXI main memory).

BTW, is the `debug interface` the same thing as the Scala source codes in `https://github.com/chipsalliance/rocket-chip/tree/master/src/main/scala/devices/debug` ? So for Rocket Chip, it does have a real hardware RTL design to load ELF into DRAM? If we tape out the Rocket Chip, we don't need some ROM to store the binary and program the ROM with a ROM programmer like what we usually do for many embedded systems?

SaltedFishLZ, 2021-08-03:
Also, Verilog supports `$readmemh` and `$writememh`. It seems we can directly manipulate memory contents in Verilog test benches (e.g., https://gitee.com/liangkangnan/tinyriscv/blob/master/tb/tinyriscv_soc_tb.v).
I wonder whether we have something similar (or simple enough like the example above) to load ELF into the CPU memory in RTL-generated emulators.

pavitrar20, 2022-11-10:
I also got this error while trying to compile a simple hello program 
riscv64-unknown-elf-gcc -march=rv64gcv hello.c -o hello          andspike pk $RISC/riscv64-unknown-elf/bin/pk hello
bbl loader
couldn't open ELF program: /home/ubuntu/Spike/riscv64-unknown-elf/bin/pk!

I tried with 32 bit variant, although I believe spike works with riscv64 
 riscv32-unknown-elf-gcc -march=rv32gcv hello.c -o hello
ubuntu@ip-172-31-17-9:~/Spike$ spike pk $RISC/riscv32-unknown-elf/bin/pk hello
bbl loader
../pk/elf.c:40: assertion failed: IS_ELF64(eh).  (Expected result?)

How to run simple programs on spike? How would the code get loaded on the simulator. 

All help is appreciated @aswaterman

pavitrar20, 2022-11-10:
I encountered the same error. How does one run the program without using spike?