riscv-software-src/riscv-tests #213: How can we see the generation of stack segment addresses as in some virtual memory tests?

sago5993, 2019-10-21:
Part I don't understand is how, a virtual memory elf test gets the sp(stack pointer) value and what's the available stack segment size?

aswaterman, 2019-10-22:
Only the supervisor code has a stack frame.  It's the 4 KiB past the `_end` symbol (see `env/v/entry.S`)

sago5993, 2019-10-22:
Thank you Andrew for the answer. Do you intend to say stack segment is between _end and _end+4KB?
what are we trying to do in below entry.S code?
handle_reset:
  la t0, trap_vector
  csrw mtvec, t0
  la sp, STACK_TOP - SIZEOF_TRAPFRAME_T

Is it a downward stack or upward?

I am very new to software side of RISCV.

aswaterman, 2019-10-22:
Yes, `_end` to `_end+4K`.

The stack grows downward by convention in RISC-V.  `STACK_TOP - SIZEOF_TRAPFRAME_T` allocates some space above the stack for the `trapframe_t` data structure, then the stack grows downward from there.

sago5993, 2019-10-23:
That explains lot of things of Andrew. What can be the reason CPU or elf execution accessing memory address _end+4096? I dont see any refernece to _end+4096 address in elf binary.

aswaterman, 2019-10-23:
If there are multiple harts, their top of stack is _end + 4K * (1 + hart id).

If there's only one hart, and you are still seeing accesses past _end + 4K, there might be a bug :)

sago5993, 2019-10-23:
Thank you Andrew for the answer. Let me look more into the issue.