riscv-software-src/riscv-tests #216: What's the purpose of setting "SLTIU" instruction?

yuanbobo345, 2019-11-06:
# What's the purpose of setting "SLTIU" instruction?

When I design my RISC-V core, I found the instruction  ***SLTIU***  always make me confused, which is described as follows:

​		***SLTI*** `(set less than immediate) places the value 1 in register rd if register rs1 is less than the sign-extended immediate when both are treated as signed numbers, else 0 is written to rd. `
​		***SLTIU*** `is similar but compares the values as unsigned numbers (i.e., the immediate is first sign-extended to XLEN bits then treated as an unsigned number). `
​		***Note***, `SLTIU rd, rs1, 1 sets rd to 1 if rs1 equals zero, otherwise sets rd to 0 (assembler pseudo instruction SEQZ rd, rs). `

Now that ***SLTIU*** compares the two value as unsigned numbers, why ***SLTIU*** instruction first sign-extends the immediate to XLEN bits?  What's the purpose of setting this instruction?

aswaterman, 2019-11-06:
It keeps the ISA simpler to have only sign-extended immediates, even when those immediates are logically unsigned quantities.

Instead of thinking of SLTIU as supporting signed immediates in the range [-2048,2047], you can think of it as supporting unsigned immediates in the union of the ranges [0,0x7ff] and [0xfffff800,0xffffffff].

yuanbobo345, 2019-11-06:
Thank you! But I think there is a missing data range between `[0x,0x7FF]` and `[0xFFFFF800,0xFFFFFFFF]` for the extended immediate. Will this make a complexity for assembly programmer?

aswaterman, 2019-11-06:
Yeah, in general, the assembly programmer must be aware of immediate range limits.  Compilers can handle this kind of thing without difficulty.

yuanbobo345, 2019-11-06:
Thank you again! I have understood the simplicity of RISC-V ISA. But I have another question: when I read the  `riscv-spec.pdf` I found the format is not straightforward, will this be changed in the next version?:smile:

aswaterman, 2019-11-06:
Which format?

yuanbobo345, 2019-11-06:
Actually, I changed the format of `riscv-spec.pdf`, but the content between `myself` and `riscv-spec.pdf` are the same. 
  The following two pictures show the difference. I think both of them has their advantages: every instruction is listed separately in my format, a series of instructions are listed in one heading in `riscv-spec.pdf` format.
[image]
[image]

allenjbaum, 2020-02-17:
I suspect that he means the immediate format, and the fact it is not a contiguous bit field, but made up of separate sub-fields that are non-contiguous and not in any specific order. As this will not be changed, and affects only the assembler, the problem is solved and this issue should be closed.

closed 2020-02-17 by aswaterman

ZhangChuann, 2022-01-25:
Why only SLTI，no SGTI ?

aswaterman, 2022-01-25:
Most potential use cases of SGTI can be synthesized by inverting the output of SLTI, which is free if the condition is used by a branch instruction.  For example,

    sgti t0, a0, 17
    beqz t0, ...

can be rewritten as

    slti t0, a0, 18
    bnez t0, ...