riscv-software-src/riscv-tests #217: Tests fail when ran in spike with "--dm-progsize 0" set by testlib.py

4ad, 2019-11-19:
Attempting to run the tests via `make debug-check` fails. 

To reproduce, inside `riscv-tests/debug` run:

    : emerald:debug; gdbserver.py targets/RISC-V/spike64.py DebugFunctionCall
    [ExamineTarget] Starting > logs/20191119-121525-spike64-ExamineTarget.log
    [ExamineTarget] pass in 1.21s
    [DebugFunctionCall] Starting > logs/20191119-121526-spike64-DebugFunctionCall.log
    [DebugFunctionCall] exception in 3.90s
    ::::::::::::::::::::::::::::[ ran 2 tests in 5s ]:::::::::::::::::::::::::::::
    1 tests returned exception
       DebugFunctionCall > logs/20191119-121526-spike64-DebugFunctionCall.log
    1 tests returned pass
    : emerald:debug; 

This test is just an arbitrary example, chosen at random, (almost) all tests fail.

Snip from the log:

    Traceback (most recent call last):
      File "/Users/aram/riscv-tests/debug/testlib.py", line 886, in run
        self.setup()
      File "./gdbserver.py", line 453, in setup
        self.gdb.load()
      File "/Users/aram/riscv-tests/debug/testlib.py", line 625, in load
        assert "failed" not in  output
    AssertionError

Further in the log we see:

    (gdb) load
    load
    Loading section .text, size 0x97c lma 0x1212340000
    Loading section .rodata, size 0x2b lma 0x1212340980
    Load failed

After further investigation, I have isolated the failure to the `--dm-progsize 0` parameter passed to spike.

What follows below is a way to reproduce the failure using a single test (any test, this one is chosen at random) without `gdbserver.py`:

In one terminal run `spike -p1 --isa RV64IMAFC --dm-auth --dm-progsize 0 --dm-sba 32 --dm-abstract-rti 30 -m0x1212340000:0x10000000 --rbb-port 9824 spike64_debug-64`,

In another, `REMOTE_BITBANG_HOST=localhost REMOTE_BITBANG_PORT=9824 WORK_AREA=0x1212340000 openocd --command 'gdb_port 52643' --command 'tcl_port disabled' --command 'telnet_port disabled' -f targets/RISC-V/spike-1.cfg`

In yet another run `riscv64-unknown-elf-gdb spike64_debug-64`

    : emerald:debug; riscv64-unknown-elf-gdb spike64_debug-64
    GNU gdb (GDB) 8.3
    Copyright (C) 2019 Free Software Foundation, Inc.
    License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
    This is free software: you are free to change and redistribute it.
    There is NO WARRANTY, to the extent permitted by law.
    Type "show copying" and "show warranty" for details.
    This GDB was configured as "--host=x86_64-apple-darwin18.7.0 --target=riscv64-unknown-elf".
    Type "show configuration" for configuration details.
    For bug reporting instructions, please see:
    <http://www.gnu.org/software/gdb/bugs/>.
    Find the GDB manual and other documentation resources online at:
        <http://www.gnu.org/software/gdb/documentation/>.
    
    For help, type "help".
    Type "apropos word" to search for commands related to "word"...
    Reading symbols from spike64_debug-64...
    (gdb) 
    (gdb) target extended-remote localhost:52643
    Remote debugging using localhost:52643
    0x000000121234095e in _exit (status=-932948524) at programs/init.c:37
    37	    while (i)
    (gdb) 
    (gdb) load
    Loading section .text, size 0x97c lma 0x1212340000
    Loading section .rodata, size 0x2b lma 0x1212340980
    Load failed
    (gdb) 

We see the same error as seen in the log.

Now try doing the same, but this time with `--dm-progsize 2` (spike default), instead of `--dm-progsize 0'.

In one terminal `spike -p1 --isa RV64IMAFC --dm-auth --dm-progsize 2 --dm-sba 32 --dm-abstract-rti 30 -m0x1212340000:0x10000000 --rbb-port 9824 spike64_debug-64`.

In another, `REMOTE_BITBANG_HOST=localhost REMOTE_BITBANG_PORT=9824 WORK_AREA=0x1212340000 openocd --command 'gdb_port 52643' --command 'tcl_port disabled' --command 'telnet_port disabled' -f targets/RISC-V/spike-1.cfg`

And gdb one more time, `riscv64-unknown-elf-gdb spike64_debug-64` 

    riscv64-unknown-elf-gdb spike64_debug-64
    GNU gdb (GDB) 8.3
    Copyright (C) 2019 Free Software Foundation, Inc.
    License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
    This is free software: you are free to change and redistribute it.
    There is NO WARRANTY, to the extent permitted by law.
    Type "show copying" and "show warranty" for details.
    This GDB was configured as "--host=x86_64-apple-darwin18.7.0 --target=riscv64-unknown-elf".
    Type "show configuration" for configuration details.
    For bug reporting instructions, please see:
    <http://www.gnu.org/software/gdb/bugs/>.
    Find the GDB manual and other documentation resources online at:
        <http://www.gnu.org/software/gdb/documentation/>.
    
    For help, type "help".
    Type "apropos word" to search for commands related to "word"...
    Reading symbols from spike64_debug-64...
    (gdb) target extended-remote localhost:52643
    Remote debugging using localhost:52643
    0x000000121234095e in _exit (status=-932948524) at programs/init.c:37
    37	    while (i)
    (gdb) 
    (gdb) load
    Loading section .text, size 0x97c lma 0x1212340000
    Loading section .rodata, size 0x2b lma 0x1212340980
    Loading section .data, size 0x2d4 lma 0x12123409b0
    Loading section .sdata, size 0x8 lma 0x1212340c88
    Start address 0x1212340000, load size 3203
    Transfer rate: 2 KB/sec, 800 bytes/write.
    (gdb) 
    (gdb) 

This time we see the load command in gdb succededing.

This patch makes the tests succeed when ran by `make debug-check`

    diff --git a/debug/testlib.py b/debug/testlib.py
    index 60cb897..e2b95a4 100644
    --- a/debug/testlib.py
    +++ b/debug/testlib.py
    @@ -123,7 +123,6 @@ class Spike(object):
             cmd += ["--dm-auth"]
     
             if not self.progbufsize is None:
    -            cmd += ["--dm-progsize", str(self.progbufsize)]
                 cmd += ["--dm-sba", "32"]
     
             if not self.dmi_rti is None:

But I do not know if this patch is correct or not.

timsifive, 2019-11-19:
We definitely want test coverage with `--dm-progsize 0`, so this is not the right way to fix it.

These tests generally expect that latest OpenOCD from https://github.com/riscv/riscv-openocd, and the latest gdb from git://sourceware.org/git/binutils-gdb.git.

Can you rerun one of these failures, adding `--server_cmd "openocd -d"` to the gdbserver.py command line, and then attach the resulting log file to this issue?

timsifive, 2019-11-19:
I think you're running into an OpenOCD bug introduced in https://github.com/riscv/riscv-openocd/commit/b7bd3f8d47b1c73d1e250a2476ee11b422494a35. I'm working on figuring this out right now.

4ad, 2019-11-20:
Unfortunately, the failure seems to be the same either at openocd HEAD, or just before that commit. I have uploaded the log.
[spike64-DebugFunctionCall.log.log](https://github.com/riscv/riscv-tests/files/3868738/spike64-DebugFunctionCall.log.log)

timsifive, 2019-11-20:
I misidentified the guilty commit. OpenOCD should work at 20804cb4d2453f959c148117bdce927feaa20fc4. I'll have some kind of fix for this today.

4ad, 2019-11-20:
I confirm that it works with that commit.

closed 2019-11-22 by timsifive

4ad, 2019-11-23:
Many thanks!