riscv-software-src/riscv-tests #228: Error when building from repository

Yi1101, 2019-12-23:
Hi,
I followed the below steps:
$ git clone https://github.com/riscv/riscv-tests
$ cd riscv-tests
$ git submodule update --init --recursive
$ autoconf
$ ./configure --prefix=$RISCV/target
$ make
$ make install

But I get error after make. The problems after running make is as below.

/home/chen/riscv/riscv-gnu-toolchain/bin/../lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccaCch9E.o: in function `.L0 ':
mm_main.c:(.text+0x20): undefined reference to `__sync_fetch_and_add_4'
collect2: error: ld returned 1 exit status
/home/chen/riscv/riscv-tests/benchmarks/Makefile:54: recipe for target 'mm.riscv' failed
make[1]: *** [mm.riscv] Error 1
make[1]: Leaving directory '/home/chen/riscv/riscv-tests/benchmarks'
Makefile:24: recipe for target 'benchmarks' failed
make: *** [benchmarks] Error 2

aswaterman, 2019-12-23:
Some tests require a compiler built with atomics support (e.g. RV64IMAFDC,
but not RV64IM). This error suggests to me that the compiler is not
assuming the A extension by default.

Yi1101, 2019-12-23:
Thanks for your reply. And I run the test one by one and some tests work

Yi1101, 2019-12-27:
However, I used the followed command to  install two compilers, and they are in the same directions. 
./configure --prefix=/home/gnu-toolchain --with-arch=rv64imc --with-abi=lp64
./configure --prefix=/home/gnu-toolchain --with-arch=rv32gc --with-abi=ilp32d
Question 1. rv32gc which includes the A extension right?
Question 2. How could I do to make the rv32gc as the compliler, in order to run  riscv-tests successfully?

Yi1101, 2019-12-27:
Please give a reply, could you?

closed 2020-03-11 by Yi1101