riscv-software-src/riscv-tests #231: run the riscv-tests

Yi1101, 2019-12-30:
When run the followed command, after running make, the error occurred as the following.
  
>$ git clone https://github.com/riscv/riscv-tests
$ cd riscv-tests
$ git submodule update --init --recursive
$ autoconf
$ ./configure --prefix=$RISCV/target
$ make
$ make install

>make[1]: riscv64-unknown-elf-gcc: Command not found
/home/chen/riscv/riscv-tests64/riscv-tests/benchmarks/Makefile:54: recipe for target 'median.riscv' failed
make[1]: *** [median.riscv] Error 127
make[1]: Leaving directory '/home/chen/riscv/riscv-tests64/riscv-tests/benchmarks'
Makefile:24: recipe for target 'benchmarks' failed
make: *** [benchmarks] Error 2

The command I used to build the gnu_ toolchain was the followed, which contained the A extension. However, the riscv-tests still has error.

>./configure --prefix=/home/chen/riscv/riscv-gnu-toolchain64/riscv-gnu-toolchain --with-arch=rv64imafdc --with-abi=ilp64d

kito-cheng, 2019-12-30:
Test script can't found your `riscv64-unknown-elf-gcc`, put toolchain install path in your PATH and try again e.g.`export PATH=/home/chen/riscv/riscv-gnu-toolchain64/riscv-gnu-toolchain/bin:$PATH`

:)

Yi1101, 2019-12-30:
Actually, I set up the path correctly, it still has the error.

closed 2020-02-18 by Yi1101