riscv-software-src/riscv-tests #234: Running virtual memory enabled programs on Spike

abdullahyildiz, 2020-01-09:
I filed this issue at https://github.com/riscv/riscv-isa-sim/issues/377 but got no response yet. I am reposting it here.

I work on Spike commit id https://github.com/riscv/riscv-isa-sim/commit/b8806095104da9c41e65696840c2863a08f82936.

Running a simulation on Spike for one of the virtual memory enabled tests at [RISC-V ISA tests](https://github.com/riscv/riscv-tests) jumps into terminate and fails (I think).

I build the test (rv64ud-v-fadd) with the following command:

`/eda/riscv/bin/riscv64-unknown-elf-gcc -march=rv64g -mabi=lp64 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -DENTROPY=0xd28d476 -std=gnu99 -O2 -I./../env/v -I./macros/scalar -T./../env/v/link.ld ./../env/v/entry.S ./../env/v/*.c rv64ud/fadd.S -o rv64ud-v-fadd
`

Then I run Spike as follows:

`spike --log-commits --isa=rv64g -l rv64ud-v-fadd`

I am trying to understand what causes the test to terminate.

I don't understand why Spike generates a page fault when executing the following command:

`80002af0:	00053007          	fld	ft0,0(a0)`

Related part of the Spike log:

```
core   0: 0x0000000000002af0 (0x00053007) fld     ft0, 0(a0)
core   0: exception trap_load_page_fault, epc 0x0000000000002af0
core   0:           tval 0x0000000000003000
core   0: 0xffffffffffe00144 (0x14011173) csrrw   sp, sscratch, sp
```

I've attached the files related to my issue.

[spike.rv64ud-v-fadd.dump.log](https://github.com/riscv/riscv-tests/files/4038982/spike.rv64ud-v-fadd.dump.log)
[rv64ud-v-fadd.dump.txt](https://github.com/riscv/riscv-tests/files/4038983/rv64ud-v-fadd.dump.txt)

abdullahyildiz, 2020-01-09:
I've observed that there are five exceptions in the related test:

1. instruction_page_fault
2. instruction_page_fault
3. load_page_fault (belongs to `test_2` at address `0x80002af0` inside the program)
4. load_page_fault (belongs to `test_2` `0x80002af0` inside the program)
5. user_ecall

At some point, the instruction `0x00053007` (`fld ft0,0(a0)`) at address `0x80002af0` does not cause a load_page_fault exception and the simulation continues.

akashtenkale03, 2023-03-02:
I've tried running different tests, and the same thing I observed is that there is always an instruction page fault after sret in pop_tf, after which a load page fault or store page fault occurs, and at some point the fault disappears and the simulation continues. Can anyone tell me whether I am on the right track?

aswaterman, 2023-03-02:
The `v` environment simulates demand paging.  Initially, none of the pages needed by the test program are mapped in.  Memory accesses cause page faults, and the trap handler brings in the relevant page.  Eventually, all of the pages needed by an instruction have been brought in, and execution can proceed.

akashtenkale03, 2023-03-02:
@aswaterman Thank you so much. Got it now

aswaterman, 2023-03-02:
@akashtenkale03 you're welcome!

closed 2023-03-02 by aswaterman