riscv-software-src/riscv-tests #241: Wrong stack pointer initialization in benchmark in multi-thread mode?

laysakura, 2020-02-08:
https://github.com/riscv/riscv-tests/blob/3a98ec2e306938cce07ab15e3678d670611aa66d/benchmarks/common/crt.S#L128-L134

Although I'm not pretty sure, I guess desired stack & TLS mapping in the 4 physical thread environment should be the following:

```text
----------- sp (core#3)
 ^
 | 2^17
 |
 v
----------- sp (core#2) = tp (core#3)
 ^
 | 2^17
 |
 v
----------- sp (core#1) = tp (core#2)
 ^
 | 2^17
 |
 v
----------- sp (core#0) = tp (core#1)
 ^
 | 2^17
 |
 v
----------- tp (core#0) = (_end + 63) & -64
```

But actually `sp` seems to have too big addresses:

```asm
sp = ((mhartid + 1) << STKSHIFT) + (mhartid << STKSHIFT) + (_end + 63) & -64
```

Stack size for `core#n` is bigger than that of `core#(n-1)`.

Am I wrong?

aswaterman, 2020-02-08:
It’s probably sloppily allocated, but does it matter? Is there actually a
bug?

laysakura, 2020-02-08:
> does it matter? Is there actually a bug?

Not really at the moment. But it's better to allocate the same size of stack + TLS for each physical core.
I will submit pull-request for this.

laysakura, 2020-02-08:
Thank you for your quick review.

closed 2020-02-08 by laysakura