riscv-software-src/riscv-tests #243: Raising exception for an illegal address

abdullahyildiz, 2020-02-12:
rv64mi-p-access test fails (as expected) on Spike with an instruction access fault:

```
core   0: 0x0000000080000140 (0x000283e7) jalr    t2, t0, 0
3 0x0000000080000140 (0x000283e7) x 7 0x0000000080000144
core   0: exception trap_instruction_access_fault, epc 0x8000000080000168
core   0:           tval 0x8000000080000168
```

I understand that this test has been written intentionally to examine whether the processor with a 64-bit PC generates an exception or not:

https://github.com/riscv/riscv-tests/blob/abfa60c8c94d40d726d2f4bd03222ac8cff585aa/isa/rv64mi/access.S#L18

I wonder that how a processor with a 40-bit PC should behave in such a situation. Should it raise an exception or not?

aswaterman, 2020-02-12:
Just because the PC is narrower than XLEN doesn't mean that the MSBs should be ignored.  You can check to see if the high bits are invalid, then summarize their validity in a single bit that gets passed down the pipeline.

abdullahyildiz, 2020-02-12:
OK. Thanks for your comment.

closed 2020-02-12 by abdullahyildiz

WadeGao, 2023-03-12:
> Just because the PC is narrower than XLEN doesn't mean that the MSBs should be ignored. You can check to see if the high bits are invalid, then summarize their validity in a single bit that gets passed down the pipeline.

Does that mean the MSB of PC must be `0` otherwise exception should be raised? I did't find the clear description about it in the spec.
It seems that a japanese netizen has the same question: https://msyksphinz.hatenablog.com/entry/2022/03/19/040000

aswaterman, 2023-03-12:
@WadeGao The PC must contain a valid address or an exception will occur.  In some systems, that might mean the MSB must be 0, but definitely not all systems.  (For example, in systems with virtual memory, it is perfectly acceptable for virtual addresses to have the MSB set, no matter which physical addresses happen to be valid.)

WadeGao, 2023-03-13:
> 

https://github.com/riscv-software-src/riscv-isa-sim/issues/540

Thank you! I also found another similar issue about this, now I understand it