riscv-software-src/riscv-tests #248: error on make: fatal error: gnu/stubs-ilp32.h: No such file or directory

snat99-hub, 2020-02-26:
I have installed the riscv-gnu-toolchain, pointed env var RISCV to that dir and did
./configure -- prefix=$RISCV/target
make.

I get the above error.

Indeed in the $RISCV/sysroot/usr/include/gnu dir I find only stubs-lp64d.h present.

I'm sorry - is the intent of these tests's build process to generate tests for all of the Arch types {rv32/rv64}{mi,si,u*} so that any can be tested to run under spike - or just one config ?

If i hack the stubs.h to define __LP64__ and __riscv_float_abi_double to get past the make problem I run into others:

make[1]: Entering directory '/mnt/c/Users/snata/Documents/tools/builds/riscv-tests/riscv-tests/isa'
riscv64-unknown-linux-gnu-gcc -march=rv32g -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -DENTROPY=0xf7930

Please advice.

aswaterman, 2020-02-26:
I think you need to build the tests with the ELF/Newlib toolchain, not the Linux/GNU one.  To do this, run `make newlib` instead of plain `make` after you run `configure`.

amoom, 2025-01-14:
you are using Linux/GNU gcc to build the test suite. but you should use the correct one, which is for ELF/Newlib. 
you can vim the Makefile under isa folder:
#--------------------------------------------------------------------
# Build rules
#--------------------------------------------------------------------

RISCV_PREFIX ?= riscv$(XLEN)-unknown-elf-
RISCV_GCC ?= $(RISCV_PREFIX)gcc
RISCV_GCC_OPTS ?= -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles
RISCV_OBJDUMP ?= $(RISCV_PREFIX)objdump --disassemble-all --disassemble-zeroes --section=.text --section=.text.startup --section=.text.init --section=.data
RISCV_SIM ?= spike

vpath %.S $(src_dir)

======================

'RISCV_GCC ?= $(RISCV_PREFIX)gcc' indicate the correct compiler you should use, it is 'riscv64-unknown-elf-gcc' .

as for your problem, I think you may need to check the riscv-gnu-toolchain. 
could you double-check whether you choose "### Installation (Newlib)".