riscv-software-src/riscv-tests #256: rv32mi-scall broken due to incorrect trap handler being used

cedric-orban, 2020-03-11:
rv32mi-scall is failing where it used to pass due to this commit: https://github.com/riscv/riscv-tests/commit/6fa1896b2a3f581359f0b6a952542f814e30602c.

This is because the test now uses the default trap handler when `__MACHINE_MODE` is defined.

aswaterman, 2020-03-11:
It seems to work for me.  Here's a successful execution trace:

```
$ spike -l --isa=rv32gc rv32mi-p-scall
core   0: 0x0000000000001000 (0x00000297) auipc   t0, 0x0
core   0: 0x0000000000001004 (0x02028593) addi    a1, t0, 32
core   0: 0x0000000000001008 (0xf1402573) csrr    a0, mhartid
core   0: 0x000000000000100c (0x0182a283) lw      t0, 24(t0)
core   0: 0x0000000000001010 (0x00028067) jr      t0
core   0: 0xffffffff80000000 (0x0480006f) j       pc + 0x48
core   0: 0xffffffff80000048 (0x00000093) li      ra, 0
core   0: 0xffffffff8000004c (0x00000113) li      sp, 0
core   0: 0xffffffff80000050 (0x00000193) li      gp, 0
core   0: 0xffffffff80000054 (0x00000213) li      tp, 0
core   0: 0xffffffff80000058 (0x00000293) li      t0, 0
core   0: 0xffffffff8000005c (0x00000313) li      t1, 0
core   0: 0xffffffff80000060 (0x00000393) li      t2, 0
core   0: 0xffffffff80000064 (0x00000413) li      s0, 0
core   0: 0xffffffff80000068 (0x00000493) li      s1, 0
core   0: 0xffffffff8000006c (0x00000513) li      a0, 0
core   0: 0xffffffff80000070 (0x00000593) li      a1, 0
core   0: 0xffffffff80000074 (0x00000613) li      a2, 0
core   0: 0xffffffff80000078 (0x00000693) li      a3, 0
core   0: 0xffffffff8000007c (0x00000713) li      a4, 0
core   0: 0xffffffff80000080 (0x00000793) li      a5, 0
core   0: 0xffffffff80000084 (0x00000813) li      a6, 0
core   0: 0xffffffff80000088 (0x00000893) li      a7, 0
core   0: 0xffffffff8000008c (0x00000913) li      s2, 0
core   0: 0xffffffff80000090 (0x00000993) li      s3, 0
core   0: 0xffffffff80000094 (0x00000a13) li      s4, 0
core   0: 0xffffffff80000098 (0x00000a93) li      s5, 0
core   0: 0xffffffff8000009c (0x00000b13) li      s6, 0
core   0: 0xffffffff800000a0 (0x00000b93) li      s7, 0
core   0: 0xffffffff800000a4 (0x00000c13) li      s8, 0
core   0: 0xffffffff800000a8 (0x00000c93) li      s9, 0
core   0: 0xffffffff800000ac (0x00000d13) li      s10, 0
core   0: 0xffffffff800000b0 (0x00000d93) li      s11, 0
core   0: 0xffffffff800000b4 (0x00000e13) li      t3, 0
core   0: 0xffffffff800000b8 (0x00000e93) li      t4, 0
core   0: 0xffffffff800000bc (0x00000f13) li      t5, 0
core   0: 0xffffffff800000c0 (0x00000f93) li      t6, 0
core   0: 0xffffffff800000c4 (0xf1402573) csrr    a0, mhartid
core   0: 0xffffffff800000c8 (0x00051063) bnez    a0, pc + 0
core   0: 0xffffffff800000cc (0x00000297) auipc   t0, 0x0
core   0: 0xffffffff800000d0 (0x01028293) addi    t0, t0, 16
core   0: 0xffffffff800000d4 (0x30529073) csrw    mtvec, t0
core   0: 0xffffffff800000d8 (0x18005073) csrwi   satp, 0
core   0: 0xffffffff800000dc (0x00000297) auipc   t0, 0x0
core   0: 0xffffffff800000e0 (0x02028293) addi    t0, t0, 32
core   0: 0xffffffff800000e4 (0x30529073) csrw    mtvec, t0
core   0: 0xffffffff800000e8 (0x800002b7) lui     t0, 0x80000
core   0: 0xffffffff800000ec (0xfff28293) addi    t0, t0, -1
core   0: 0xffffffff800000f0 (0x3b029073) csrw    pmpaddr0, t0
core   0: 0xffffffff800000f4 (0x01f00293) li      t0, 31
core   0: 0xffffffff800000f8 (0x3a029073) csrw    pmpcfg0, t0
core   0: 0xffffffff800000fc (0x00000297) auipc   t0, 0x0
core   0: 0xffffffff80000100 (0x01828293) addi    t0, t0, 24
core   0: 0xffffffff80000104 (0x30529073) csrw    mtvec, t0
core   0: 0xffffffff80000108 (0x30205073) csrwi   medeleg, 0
core   0: 0xffffffff8000010c (0x30305073) csrwi   mideleg, 0
core   0: 0xffffffff80000110 (0x30405073) csrwi   mie, 0
core   0: 0xffffffff80000114 (0x00000193) li      gp, 0
core   0: 0xffffffff80000118 (0x00000297) auipc   t0, 0x0
core   0: 0xffffffff8000011c (0xeec28293) addi    t0, t0, -276
core   0: 0xffffffff80000120 (0x30529073) csrw    mtvec, t0
core   0: 0xffffffff80000124 (0x00100513) li      a0, 1
core   0: 0xffffffff80000128 (0x01f51513) slli    a0, a0, 31
core   0: 0xffffffff8000012c (0x00054c63) bltz    a0, pc + 24
core   0: 0xffffffff80000144 (0x00000293) li      t0, 0
core   0: 0xffffffff80000148 (0x00028a63) beqz    t0, pc + 20
core   0: 0xffffffff8000015c (0x30005073) csrwi   mstatus, 0
core   0: 0xffffffff80000160 (0x00002537) lui     a0, 0x2
core   0: 0xffffffff80000164 (0x80050513) addi    a0, a0, -2048
core   0: 0xffffffff80000168 (0x30052073) csrs    mstatus, a0
core   0: 0xffffffff8000016c (0x00000297) auipc   t0, 0x0
core   0: 0xffffffff80000170 (0x01428293) addi    t0, t0, 20
core   0: 0xffffffff80000174 (0x34129073) csrw    mepc, t0
core   0: 0xffffffff80000178 (0xf1402573) csrr    a0, mhartid
core   0: 0xffffffff8000017c (0x30200073) mret
core   0: 0x0000000080000180 (0x00200193) li      gp, 2
core   0: 0xffffffff80000184 (0x00800313) li      t1, 8
core   0: 0xffffffff80000188 (0x000022b7) lui     t0, 0x2
core   0: 0xffffffff8000018c (0x80028293) addi    t0, t0, -2048
core   0: 0xffffffff80000190 (0x3002b073) csrc    mstatus, t0
core   0: 0xffffffff80000194 (0x300023f3) csrr    t2, mstatus
core   0: 0xffffffff80000198 (0x0072f2b3) and     t0, t0, t2
core   0: 0xffffffff8000019c (0x00028463) beqz    t0, pc + 8
core   0: 0xffffffff800001a4 (0x000022b7) lui     t0, 0x2
core   0: 0xffffffff800001a8 (0x80028293) addi    t0, t0, -2048
core   0: 0xffffffff800001ac (0x3002b073) csrc    mstatus, t0
core   0: 0xffffffff800001b0 (0x00000297) auipc   t0, 0x0
core   0: 0xffffffff800001b4 (0x01028293) addi    t0, t0, 16
core   0: 0xffffffff800001b8 (0x34129073) csrw    mepc, t0
core   0: 0xffffffff800001bc (0x30200073) mret
core   0: 0x00000000800001c0 (0x00100193) li      gp, 1
core   0: 0xffffffff800001c4 (0x00000073) ecall
core   0: exception trap_user_ecall, epc 0xffffffff800001c4
core   0: 0x0000000080000004 (0x34202f73) csrr    t5, mcause
core   0: 0xffffffff80000008 (0x00800f93) li      t6, 8
core   0: 0xffffffff8000000c (0x03ff0863) beq     t5, t6, pc + 48
core   0: 0xffffffff8000003c (0x00001f17) auipc   t5, 0x1
core   0: 0xffffffff80000040 (0xfc3f2223) sw      gp, -60(t5)
core   0: 0xffffffff80000044 (0xff9ff06f) j       pc - 0x8
```

aswaterman, 2020-03-11:
I see the issue is that with different test environments, this can fail.  I'll just revert it to avoid belaboring the point.

commit ee576c08 (2020-03-11): Revert "scall: make the intention of the test in machine mode more clear (#246)"

This reverts commit 6fa1896b2a3f581359f0b6a952542f814e30602c.

Resolves #256
closed 2020-03-11 by aswaterman