riscv-software-src/riscv-tests #282: build error - multiple definitions of 'tohost' and 'fromhost'

PICOVERAVR, 2020-06-13:
I'm trying to build the risc-v test suite, and I'm getting a lot of build errors saying something like `/home/kneil/code/lmisc/riscv-tools/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/cceYUWQN.o: in function 'tohost':
(.tohost+0x0): multiple definition of 'tohost'; /tmp/cc6ycU4L.o:(.sbss+0x10): first defined here
/home/kneil/code/lmisc/riscv-tools/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/cceYUWQN.o: in function 'fromhost':
(.tohost+0x40): multiple definition of 'fromhost'; /tmp/cc6ycU4L.o:(.sbss+0x8): first defined here`.

I have the riscv-gnu-toolchain (with the default target) installed and $RISCV set to `/home/kneil/code/lmisc/riscv-tools/bin`, and the build command I am using is `make clean && ./configure --prefix=$RISCV/rv32 && make -j12`.  I tried using `make` instead of `make -j12`, and that did not fix the problem.

(I also changed the backticks in the output from gcc to quotes so that the code is formatted properly)

Daxidz, 2020-06-22:
I got the same issue. Using `riscv64-unknown-elf-gcc (GCC) 10.1.0` and compiling from my host:

```
riscv64-unknown-elf-gcc -march=rv32g -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -DENTROPY=0xf7930f7 -std=gnu99 -O2 -I/home/david/disk2/Projects/RISCAI/riscv-tests/isa/../env/v -I/home/david/disk2/Projects/RISCAI/riscv-tests/isa/macros/scalar -T/home/david/disk2/Projects/RISCAI/riscv-tests/isa/../env/v/link.ld /home/david/disk2/Projects/RISCAI/riscv-tests/isa/../env/v/entry.S /home/david/disk2/Projects/RISCAI/riscv-tests/isa/../env/v/*.c rv32ui/simple.S -o rv32ui-v-simple
/opt/riscv64/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccPaSiZy.o: in function `tohost':
(.tohost+0x0): multiple definition of `tohost'; /tmp/ccfJGtzb.o:(.sbss+0x10): first defined here
/opt/riscv64/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccPaSiZy.o: in function `fromhost':
(.tohost+0x40): multiple definition of `fromhost'; /tmp/ccfJGtzb.o:(.sbss+0x8): first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [/home/david/disk2/Projects/RISCAI/riscv-tests/isa/Makefile:74: rv32ui-v-simple] Error 1
make[1]: Leaving directory '/home/david/disk2/Projects/RISCAI/riscv-tests/isa'
make: *** [Makefile:29: isa] Error 2

```

Daxidz, 2020-06-22:
Got it to compile using the precompiled toolchain from SciFive found [here](https://www.sifive.com/boards):
```
$ riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (SiFive GCC 8.3.0-2019.08.0) 8.3.0

```

PICOVERAVR, 2020-06-23:
Compiled with zero errors.  Thank you!

closed 2020-06-23 by PICOVERAVR