riscv-software-src/riscv-tests #286: Erro while building - Multiple definitions of `tohost` and `fromhost`

pcotret, 2020-07-14:
- Installed the newlib cross-compiler: https://github.com/riscv/riscv-gnu-toolchain
- Added the compiler to my PATH
- export RISCV=<toolchain-directory>
I got an error while compiling `riscv-tests`:
```bash
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccJxHjmW.o: in function `tohost':
(.tohost+0x0): multiple definition of `tohost'; /tmp/ccZrk0sX.o:(.sbss+0x10): first defined here
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccJxHjmW.o: in function `fromhost':
(.tohost+0x40): multiple definition of `fromhost'; /tmp/ccZrk0sX.o:(.sbss+0x8): first defined here
collect2: error: ld returned 1 exit status
```
The solution provided in #282 isn't satisfying for me as I would like to use my own compiler instead of downloading binaries from a third-party.

Any idea?

aswaterman, 2020-07-14:
Recent versions of GCC default to `-fno-common`, causing these multiple-definition errors.  I didn't see them because I'm still using an older version of GCC.  This is now fixed on master.

closed 2020-07-14 by aswaterman

pcotret, 2020-07-14:
Thanks @aswaterman !

zhutang, 2021-02-24:
> * Installed the newlib cross-compiler: https://github.com/riscv/riscv-gnu-toolchain
> * Added the compiler to my PATH
> * `export RISCV=<toolchain-directory>`
>   I got an error while compiling `riscv-tests`:
> 
> ```shell
> /opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccJxHjmW.o: in function `tohost':
> (.tohost+0x0): multiple definition of `tohost'; /tmp/ccZrk0sX.o:(.sbss+0x10): first defined here
> /opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccJxHjmW.o: in function `fromhost':
> (.tohost+0x40): multiple definition of `fromhost'; /tmp/ccZrk0sX.o:(.sbss+0x8): first defined here
> collect2: error: ld returned 1 exit status
> ```
> 
> The solution provided in #282 isn't satisfying for me as I would like to use my own compiler instead of downloading binaries from a third-party.
> 
> Any idea?

Excuse me, did you use your own compiler to fix this bug. I just git the master code of riscv-gnu-toolchain and compile them, and I want to test them for the opcodes-custom. however, i got this bug too.

MstMoonshine, 2021-03-17:
> Recent versions of GCC default to `-fno-common`, causing these multiple-definition errors. I didn't see them because I'm still using an older version of GCC. This is now fixed on master.

Thanks, man. I solved this by adding `-fcommon` flag to Makefile under ./isa directory, following your hint.

commit 3261a310 (2020-07-14): bump env to fix #286