riscv-software-src/riscv-tests #287: Failing debug test case "spike32.SimpleV13Test"

JanMatCodasip, 2020-08-05:
This debug test is failing on my machine: "spike32.SimpleV13Test":

```bash
cd debug 
make run.spike32.SimpleV13Test
```

I am attaching my log: [20200805-104914-spike32-SimpleV13Test.log](https://github.com/riscv/riscv-tests/files/5027266/20200805-104914-spike32-SimpleV13Test.log)

Have you please seen this issue already or does it look like a problem in my setup?

After further digging, it turns out that the test always fails when reading back the value of `$v13.b[2]` field (https://github.com/riscv/riscv-tests/blob/master/debug/gdbserver.py#L129). Reading the preceding fields `$v13.b[0]` and `$v13.b[1]` never failed. Dump of the whole register v13 prior to testing the individual fields (https://github.com/riscv/riscv-tests/blob/master/debug/gdbserver.py#L127) shows that the values are already different than what was written to them. See also the log file.

I am running the tests on a Debian 9 machine.

These are the software components I am using (built from source):
* riscv-tests: latest commit as of now - 3261a31
* riscv-gnu-toolchain: 69fd36ba (Jul 29)
* spike: 6275cdf0 (Jul 28)
* riscv-openocd: 9ed60777 (Jul 17)

Thank you, 

Jan

JanMatCodasip, 2020-08-05:
Further information - 

This test `SimpleV13Test` fails not only on `spike32` but also on `spike32-2` and `spike64`:

```
$ cat results.txt | grep -C 5 "returned fail"
Result: fail
Logfile: logs/20200805-133024-spike32-SimpleV13Test.log
Reproduce: ./gdbserver.py ./targets/RISC-V/spike32.py SimpleV13Test
Time elapsed: 10.16s
::::::::::::::::::::::::::::[ ran 1 tests in 10s ]::::::::::::::::::::::::::::
1 tests returned fail
   SimpleV13Test > logs/20200805-133024-spike32-SimpleV13Test.log
echo run.spike32.SmpSimultaneousRunHalt
run.spike32.SmpSimultaneousRunHalt
./gdbserver.py \
    ./targets/RISC-V/spike32.py \
--
Result: fail
Logfile: logs/20200805-134244-spike32_2-SimpleV13Test.log
Reproduce: ./gdbserver.py ./targets/RISC-V/spike32-2-hwthread.py SimpleV13Test
Time elapsed: 16.49s
::::::::::::::::::::::::::::[ ran 1 tests in 16s ]::::::::::::::::::::::::::::
1 tests returned fail
   SimpleV13Test > logs/20200805-134244-spike32_2-SimpleV13Test.log
echo run.spike32-2-hwthread.SmpSimultaneousRunHalt
run.spike32-2-hwthread.SmpSimultaneousRunHalt
./gdbserver.py \
    ./targets/RISC-V/spike32-2-hwthread.py \
--
Result: fail
Logfile: logs/20200805-140057-spike64_2-SimpleV13Test.log
Reproduce: ./gdbserver.py ./targets/RISC-V/spike64-2.py SimpleV13Test
Time elapsed: 103.17s
:::::::::::::::::::::::::::[ ran 1 tests in 103s ]::::::::::::::::::::::::::::
1 tests returned fail
   SimpleV13Test > logs/20200805-140057-spike64_2-SimpleV13Test.log
echo run.spike64-2.SmpSimultaneousRunHalt
run.spike64-2.SmpSimultaneousRunHalt
./gdbserver.py \
    ./targets/RISC-V/spike64-2.py \
```

Jan

JanMatCodasip, 2020-08-05:
As a side note, another 180 tests fail (exception) due to the recent deprecation of "-rtos riscv" option in OpenOCD (https://github.com/riscv/riscv-openocd/pull/499).

timsifive, 2020-08-05:
It used to work. I guess something changed in spike's vector implementation, which may have been because the vector spec changed. I'll do some digging.

timsifive, 2020-08-05:
This broke in spike commit 308b6db54946d853b47cb5085a06e90aae14145c, which is mercifully short.

timsifive, 2020-08-05:
This patch fixes it in OpenOCD. I need to figure out if it's important for OpenOCD to support 0.9 before I just merge that:
```
diff --git a/src/target/riscv/riscv-013.c b/src/target/riscv/riscv-013.c
index 2ea8fdff6..cd29f3106 100644
--- a/src/target/riscv/riscv-013.c
+++ b/src/target/riscv/riscv-013.c
@@ -1875,7 +1875,7 @@ static int prep_for_vector_access(struct target *target, uint64_t *vtype,
        if (register_read(target, vl, GDB_REGNO_VL) != ERROR_OK)
                return ERROR_FAIL;
 
-       if (register_write_direct(target, GDB_REGNO_VTYPE, encoded_vsew << 2) != ERROR_OK)
+       if (register_write_direct(target, GDB_REGNO_VTYPE, encoded_vsew << 3) != ERROR_OK)
                return ERROR_FAIL;
        *debug_vl = DIV_ROUND_UP(r->vlenb[r->current_hartid] * 8,
                        riscv_xlen(target));
```

JanMatCodasip, 2020-08-06:
Thanks, Tim.

After applying the two changes above ("enable_rtos_riscv" + the OpenOCD patch), all tests pass for me:

```bash
$ cd debug

$ make all 2>&1 | tee results.txt

$ cat results.txt | grep "tests returned" | sort | uniq -c
     56 1 tests returned not_applicable
    384 1 tests returned pass

```

closed 2020-08-06 by JanMatCodasip