riscv-software-src/riscv-tests #289: Debug tests fail with latest tools

sobuch, 2020-08-11:
Tests pass for:
riscv-openocd: e2d3184e + patch from https://github.com/riscv/riscv-tests/issues/287#issuecomment-669279248
riscv-tests: f68760e7
riscv-gnu-toolchain: 69fd36ba
riscv-isa-sim: 6859ccfa

However, if either riscv-gnu-toolchain or riscv-isa-sim is updated to the later version (aba677ba, da34b0ee respectively) some tests will fail.

In case of riscv-gnu-toolchain the issue is probably caused by the updated version of gnu-binutils, only when V extension is used in march. Issue presents itself by the following error during compilation:
```
+ riscv64-unknown-elf-gcc -g programs/checksum.c programs/tiny-malloc.c programs/infinite_loop.S -DDEFINE_MALLOC -DDEFINE_FREE programs/entry.S programs/init.c -DNHARTS=1 -I ../env -T targets/RISC-V/spike32.lds -nostartfiles -mcmodel=medany -DXLEN=32 -o /tmp/spike32_checksum-32_4nqkw5ui -march=rv32imafdcv -mabi=ilp32
riscv64-unknown-elf/bin/ld: error: /tmp/ccqlBLYz.o: Mis-matched ISA version for 'm' extension. 2.0 vs 2.0
riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /tmp/ccqlBLYz.o
riscv64-unknown-elf/bin/ld: -march=: ISA string must begin with rv32 or rv64
riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /tmp/ccd8hdaj.o
riscv64-unknown-elf/bin/ld: -march=: ISA string must begin with rv32 or rv64
riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /tmp/cceWHEo2.o
collect2: error: ld returned 1 exit status
```

Log from the second issue, encountered when updating spike, is attached here:
[20200810-141746-spike64_2-Sv39Test.log](https://github.com/riscv/riscv-tests/files/5051493/20200810-141746-spike64_2-Sv39Test.log)

kito-cheng, 2020-08-11:
I would suggest you using rvv branch for riscv-gnu-toolchain if you want to testing `V` extension.

sobuch, 2020-08-11:
Thanks for the suggestion but I was not looking specifically to test the V extension. Maybe it could be useful to include also some (latest) working revisions for requirements in the readme?

sobuch, 2020-08-14:
I wanted to try the `rvv-0.9.x` branch before closing this issue, but was unable to get it to build. I am getting the following error:
```
riscv-gnu-toolchain/build/../riscv-binutils/bfd/elfxx-riscv.c: In function ‘riscv_multi_letter_ext_valid_p’:
riscv-gnu-toolchain/build/../riscv-binutils/bfd/elfxx-riscv.c:1404:3: error: ‘for’ loop initial declarations are only allowed in C99 mode
   for (size_t i = 0; known_exts[i]; ++i)
   ^
riscv-gnu-toolchain/build/../riscv-binutils/bfd/elfxx-riscv.c:1404:3: note: use option -std=c99 or -std=gnu99 to compile your code
riscv-gnu-toolchain/build/../riscv-binutils/bfd/elfxx-riscv.c: In function ‘riscv_parse_subset’:
riscv-gnu-toolchain/build/../riscv-binutils/bfd/elfxx-riscv.c:1493:3: error: ‘for’ loop initial declarations are only allowed in C99 mode
   for (size_t i = 0; i < ARRAY_SIZE (parse_config); ++i) {
```
After the first failure, I cloned a new repository so I am pretty sure there is no problem with old submodules or anything from a previous build. Tried this in two different environments, also if I change the branch to master and update submodules it succeeds. So I really have no idea where the problem could be. These were my steps:
```
git clone https://github.com/riscv/riscv-gnu-toolchain -b rvv-0.9.x
cd riscv-gnu-toolchain
git submodule update --init --recursive
mkdir build && cd build
export RISCV=`pwd`
../configure --prefix=$RISCV
make
```
Could you help me with this?