riscv-software-src/riscv-tests #292: Question about rv64uc-p-rvc test

knute-sifive, 2020-08-26:
While stepping through this test on Spike, I'm baffled on the behavior I'm observing.  In the test code (rvc.S), there's this comment:
>   // Make sure fetching a 4-byte instruction across a page boundary works.

Not sure this _should_ work based on my initial read of the Spec, section 2.2:
>  All [instruction formats] are a fixed 32 bits in length and must be aligned on a four-byte boundary in memory. An instruction-address-misaligned exception is generated on a taken branch or unconditional jump if the target address is not four-byte aligned.

But an addendum states (in the same section) reads:
> The alignment constraint for base ISA instructions is relaxed to a two-byte boundary when instruction extensions with 16-bit lengths or other odd multiples of 16-bit lengths are added (i.e., IALIGN=16).

Indeed this test (being compressed) falls into the exception (no pun) and not the norm.

But the behavior on Spike is still confusing.  Here's the output from Spike:

```
core   0: 0x0000000080000180 (0x67f0106f) j       pc + 0x1e7e
core   0: 0x0000000080001ffe (0x00158593) addi    a1, a1, 1
core   0: 0x0000000080002002 (0x29b00393) li      t2, 667
```
Note the jump to `0x1ffe`, which is misaligned (to a 4K page).  Spike fetches and decodes 4 bytes at that address.  Assuming a 4K page, is this the expected behavior?   To me, this is a potential bus error waiting to happen.

aswaterman, 2020-08-26:
Yes, if IALIGN=16--as is the case when the C extension is implemented--then instructions may begin on any 2-byte boundary.  Implementations with IALIGN=16 need to cope with that microarchitecturally, e.g., by performing two instruction fetches and combining them, or using a banked instruction memory.

If you're only implementing e.g. RV32I, then IALIGN=32, and attempting to fetch 0x1ffe will raise an instruction-address-misaligned exception.

knute-sifive, 2020-08-26:
> e.g., by performing two instruction fetches and combining them, or using a banked instruction memory.

Ah, that's the key I was missing.  This would definitely allow for misalignment memory errors for the second access.  Good.  Thank you.

closed 2020-08-26 by knute-sifive